Robot-Will/Stino #227: Add Custom Library Path ?

hallard, 2015-01-06:
Hi,

I discovered Sublime Text and Stino, whooo wonderfull work, very nice and useful ;-)

Since you plugins seems to totally setup the parameters on command line for compilation, I'm wondering if it would be simple to add another custom library path (keeping the default working also) ?

I'm asking this because I'm working on different Arduino IDE version (1.0.5; 1.0.6, 1.5.8) and each time I need to manage (copy/paste) my custom libraries from one environment to other each time I'm modify my lib so having a custom fixed library path could be really great to have only one version of my libs. But may be it could work with "extra flags" feature ? 

Hoping you find this idea useful.

xseignard, 2015-01-13:
+1000!

would be a great addition :+1:

T3KNOCRAT, 2015-10-31:
Yes, please!

closed 2016-01-28 by gepd

nodemand, 2023-01-31:
Me tooo like a lot!