Robot-Will/Stino #228: Servo.h compilation error

soragon, 2015-01-08:
Whenever I include <Servo.h> in Stino, the compiler throws back an error, when it compiles fine in Arduino 1.5.7.

```
/Applications/Arduino.app/Contents/Java/libraries/Servo/src/Servo.h:67:2: error: #error "This library only supports boards with an AVR or SAM processor."
#error "This library only supports boards with an AVR or SAM processor."
```

Other people are having [similar issues](https://translate.google.com.au/translate?sl=auto&tl=en&js=y&prev=_t&hl=en&ie=UTF-8&u=http%3A%2F%2Fforum.arduino.cc%2Findex.php%3Ftopic%3D274906.0&edit-text=&act=url).

Any ideas on how to fix this?

gonzoid, 2015-01-09:
Hello @soragon 

Which target platform/board did you select for compilation in the `Arduino > Board` menu?

I tried the included example sketches `Knob` and `Sweep` for `Arduino Uno` board (AVR) and it buildt fine:
```
[Stino - Start building "Sweep"...]
[  3%] Creating C:\Users\Gonzo\Documents\Arduino_Build\Sweep\Sweep.ino.cpp.o...
[  6%] Creating C:\Users\Gonzo\Documents\Arduino_Build\Sweep\lib_Servo\src\avr\Servo.cpp.o...
[...snip...]
[100%] Creating C:\Users\Gonzo\Documents\Arduino_Build\Sweep\Sweep.hex...
Sketch uses 2,564 bytes (7.9%) of program storage space. Maximum is 32,256 bytes.
Global variables use 56 bytes (2.7%) of dynamic memory, leaving 1,992 bytes for local variables. Maximum is 2,048 bytes.
[Stino - Done building "Sweep" in 4.1s.]
```

But it didn't for `Arduino Due` board (ARM/SAM), I got the same error for `Programming Port` and `Native USB Port` variants, **for another reason though**:
```
[Stino - Start building "Sweep"...]
[  3%] Creating C:\Users\Gonzo\Documents\Arduino_Build\Sweep\Sweep.ino.cpp.o...
[  6%] Creating C:\Users\Gonzo\Documents\Arduino_Build\Sweep\lib_Servo\src\sam\Servo.cpp.o...
[...snip...]
[ 87%] Creating C:\Users\Gonzo\Documents\Arduino_Build\Sweep\core_arduino\USB\USBCore.cpp.o...
<command-line>:0:13: error: 'Arduino' was not declared in this scope

D:/Program Files/Arduino-1.5.8/hardware/arduino/sam/cores/arduino/USB/USBCore.cpp:69:34: note: in expansion of macro 'USB_PRODUCT'

 const uint8_t STRING_PRODUCT[] = USB_PRODUCT;
                                  ^

[Stino - Exit with error code 1.]
```

It seems a constant is not defined properly in your case. See `libraries/Servo/src/Servo.h`, line 67 as your error output point out:
``` c++
// Architecture specific include
#if defined(ARDUINO_ARCH_AVR)
#include "avr/ServoTimers.h"
#elif defined(ARDUINO_ARCH_SAM)
#include "sam/ServoTimers.h"
#else
#error "This library only supports boards with an AVR or SAM processor."
#endif
```

Could you update your Arduino app to 1.5.8 and try again to build on both Arduino and Stino please?

gonzoid, 2015-01-09:
You can set verbose compiler output in the menu `Arduino > Build Options > Show Compilation Output`.

Here is a snippet of what I get, in my case the constant is defined as expected with `-DARDUINO_ARCH_AVR`.
```
[Stino - Start building "Sweep"...]
[  3%] Creating C:\Users\Gonzo\Documents\Arduino_Build\Sweep\Sweep.ino.cpp.o...
"D:\Program Files\Arduino-1.5.8/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD  -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=158 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-ID:\Program Files\Arduino-1.5.8\libraries\Servo\examples\Sweep" "-ID:\Program Files\Arduino-1.5.8\hardware\arduino\avr\cores\arduino" "-ID:\Program Files\Arduino-1.5.8\hardware\arduino\avr\variants\standard" "-ID:\Program Files\Arduino-1.5.8\libraries\Servo\src" "-ID:\Program Files\Arduino-1.5.8\libraries\Servo\src\avr" "C:\Users\Gonzo\Documents\Arduino_Build\Sweep\Sweep.ino.cpp" -o "C:\Users\Gonzo\Documents\Arduino_Build\Sweep\Sweep.ino.cpp.o"
[...snip...]
```

soragon, 2015-01-10:
Sure thing! I tried to compile under Arduino Uno (AVR).

In Stino while still on 1.5.7:

`````````````````
Compiling Desktop...
Creating /Users/Michel/Documents/Arduino_Build/Desktop/Desktop.ino.cpp.o...
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=157 -DARDUINO_AVR_UNO -DARDUINO_ARCH_   "-I/Users/Michel/Documents/Arduino_Build/Desktop" "-I/Users/Michel/Desktop" "-I/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "-I/Applications/Arduino.app/Contents/Java/libraries/Servo/src" "-I/Applications/Arduino.app/Contents/Java/libraries/Servo/src/avr" "-I/Applications/Arduino.app/Contents/Java/libraries/Servo/src/sam"  "/Users/Michel/Documents/Arduino_Build/Desktop/Desktop.ino.cpp" -o "/Users/Michel/Documents/Arduino_Build/Desktop/Desktop.ino.cpp.o"
In file included from /Users/Michel/Documents/Arduino_Build/Desktop/Desktop.ino.cpp:3:0:
/Applications/Arduino.app/Contents/Java/libraries/Servo/src/Servo.h:67:2: error: #error "This library only supports boards with an AVR or SAM processor."
 #error "This library only supports boards with an AVR or SAM processor."
  ^
[Stino - Error 1]
`````````````````

In Arduino IDE 1.5.7:
`````````````````
Using library Servo in folder: /Applications/Arduino.app/Contents/Java/libraries/Servo 

/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=157 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard -I/Applications/Arduino.app/Contents/Java/libraries/Servo/src /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/test.cpp -o /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/test.cpp.o 
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/Servo/avr/Servo.cpp.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/Servo/sam/Servo.cpp.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/hooks.c.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/WInterrupts.c.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/wiring.c.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/wiring_analog.c.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/wiring_digital.c.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/wiring_pulse.c.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/wiring_shift.c.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/CDC.cpp.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/HardwareSerial.cpp.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/HardwareSerial0.cpp.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/HardwareSerial1.cpp.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/HardwareSerial2.cpp.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/HardwareSerial3.cpp.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/HID.cpp.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/IPAddress.cpp.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/main.cpp.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/new.cpp.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/Print.cpp.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/Stream.cpp.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/Tone.cpp.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/USBCore.cpp.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/WMath.cpp.o
Using previously compiled file: /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build3123054743046774233.tmp/WString.cpp.o
[... cut ...]
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-objcopy -O ihex -R .eeprom /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build8608701785134039810.tmp/test.cpp.elf /var/folders/w_/ttn4n5td227_bkjlj573m3wr0000gn/T/build8608701785134039810.tmp/test.cpp.hex 

Sketch uses 1,112 bytes (3%) of program storage space. Maximum is 32,256 bytes.
Global variables use 51 bytes (2%) of dynamic memory, leaving 1,997 bytes for local variables. Maximum is 2,048 bytes.
```````````````````

Kind of strange that it's only defined as `-DARDUINO_ARCH_`. Why would this be the case? See image below...
[image]

gonzoid, 2015-01-10:
Looking at your menu screenshot, I think the actual problem is your Stino version.

Stino's author is now working on a new branch/version and the older which you are using (probably from ST's Package Control) is no longer supported. This new version of Stino will hit Package Control once it'll be stable enough.

You can find a way to manually install the last version here: https://github.com/Robot-Will/Stino/issues/179#issuecomment-61121785

soragon, 2015-01-11:
Ah you're right! All working with the latest version. Thanks!

closed 2015-01-11 by soragon