Robot-Will/Stino #229: "Error loading meta info from..."

PeterBeklemishev, 2015-01-11:
I installed ST2, package control and unzip downloaded Stino-new-stino.zip to `/home/peter/.config/sublime-text-2/Packages`

Sublime show me an messagebox, which contain:

```
Error loading meta info from Packages/Stino-new-stino/Comments.tmPreferences: Error parsing plist xml: Failed to open file In file "Packages/Stino-new-stino/Comments.tmPreferences"
```

in the ST console:
```
startup, version: 2221 linux x64 channel: stable
executable: /home/peter/Загрузки/Sublime Text 2/sublime_text
working dir: /home/peter/Загрузки
packages path: /home/peter/.config/sublime-text-2/Packages
settings path: /home/peter/.config/sublime-text-2/Settings
PackageSetup not required
Py_GetProgramName(): sublime_text
Py_GetExecPrefix(): .
Py_GetProgramFullPath(): 
Py_GetPath(): ./lib/python26.zip:./lib/python2.6/:./lib/python2.6/plat-linux2:./lib/python2.6/lib-tk:./lib/python2.6/lib-old:./lib/python2.6/lib-dynload
Py_GetPythonHome(): .
catalogue loaded
found 5 files for base name Default.sublime-keymap
found 1 files for base name Default.sublime-mousemap
found 2 files for base name Main.sublime-menu
Generating syntax summary
Generating meta info summary
Reloading plugin /home/peter/.config/sublime-text-2/Packages/0_package_control_loader/00-package_control.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/0_package_control_loader/01-ssl-linux.py
Linux SSL: successfully loaded _ssl module for libssl.so.1.0.0
Reloading plugin /home/peter/.config/sublime-text-2/Packages/0_package_control_loader/02-bz2.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/CSS/css_completions.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/comment.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/copy_path.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/delete_word.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/detect_indentation.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/duplicate_line.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/echo.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/exec.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/fold.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/font.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/goto_line.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/indentation.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/kill_ring.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/mark.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/new_templates.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/open_file_settings.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/open_in_browser.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/paragraph.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/save_on_focus_lost.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/scroll.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/set_unsaved_view_name.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/side_bar.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/sort.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/swap_line.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/switch_file.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/transform.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/transpose.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Default/trim_trailing_white_space.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Diff/diff.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/HTML/encode_html_entities.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/HTML/html_completions.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Package Control/Package Control.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Package Control/bootstrap.py
Reloading plugin /home/peter/.config/sublime-text-2/Packages/Stino-new-stino/StinoStarter.py
plugin init time: 0.283208
loading bindings
loading pointer bindings
Exception in thread Thread-2:
Traceback (most recent call last):
  File ".\threading.py", line 532, in __bootstrap_inner
  File ".\threading.py", line 484, in run
  File "./stino/pyarduino/base/serial_listener.py", line 44, in update
  File "./stino/main.py", line 500, in update_serial_info
AttributeError: 'NoneType' object has no attribute 'active_view'

found 1 files for base name Default.sublime-theme
theme loaded
app ready
pre session restore time: 0.406533
using gamma: 1 (err: 0)
wrote startup cache, added files: 3 orphaned files: 11 total files: 97 cache hits: 94
startup time: 0.558965 (package setup was not run)
found 3 files for base name Main.sublime-menu
Generating syntax summary
found 4 files for base name Main.sublime-menu
found 5 files for base name Main.sublime-menu
found 6 files for base name Main.sublime-menu
found 7 files for base name Main.sublime-menu
found 8 files for base name Main.sublime-menu
found 9 files for base name Main.sublime-menu
found 10 files for base name Main.sublime-menu
found 11 files for base name Main.sublime-menu
found 12 files for base name Main.sublime-menu
loaded 901 snippets
Package Control: Removed directory for orphaned package Stino-new-stino
Package Control: Skipping automatic upgrade, last run at 2015-01-11 20:38:18, next run at 2015-01-11 21:38:18 or after
Generating meta info summary
error: Error loading meta info from Packages/Stino-new-stino/Comments.tmPreferences: Error parsing plist xml: Failed to open file In file "Packages/Stino-new-stino/Comments.tmPreferences"
Unloading plugin /home/peter/.config/sublime-text-2/Packages/Stino-new-stino/StinoStarter.py
found 4 files for base name Default.sublime-keymap
```

I'm on:
```
peter@peter-Aspire-V3-571G:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.1 LTS
Release:	14.04
Codename:	trusty
peter@peter-Aspire-V3-571G:~$ uname -a
Linux peter-Aspire-V3-571G 3.13.0-43-generic #72-Ubuntu SMP Mon Dec 8 19:35:06 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux
peter@peter-Aspire-V3-571G:~$ 
```

gonzoid, 2015-01-11:
Hello @PeterBeklemishev 

Apparently this problem happens on a lot of different ST plugins, and Package Control seems to be the cause.
You could try some of the methods from http://www.sublimetext.com/forum/viewtopic.php?f=3&t=8210

gonzoid, 2015-01-11:
I updated my version of Stino and got the same problem, this can be resolved as per #221

PeterBeklemishev, 2015-01-12:
@gonzoid thank you, that fix it!

closed 2015-01-12 by PeterBeklemishev