Robot-Will/Stino #230: Comments can break compilation

aMoniker, 2015-01-17:
Ran into this issue today, something Stino is doing seems to interpret comments as code, breaking compilation in certain cases. This does *not* happen in the normal Arduino IDE.

This code works:
```
void setup() {
  pinMode(13, OUTPUT);
}

void loop() {
  digitalWrite(13, HIGH);
  delay(100);
  digitalWrite(13, LOW);
  delay(100);
}
```

But this code will not compile:

```
// void setup()

void setup() {
  pinMode(13, OUTPUT);
}

void loop() {
  digitalWrite(13, HIGH);
  delay(100);
  digitalWrite(13, LOW);
  delay(100);
}
```

So, having that commented `// void setup()` will break compilation.

raymusmunt, 2015-01-17:
Stino doesn't like commented out setups on my machine either.

gro00, 2015-01-18:
How about changing line ending option from "Mac OS" to "Windows" or "Unix"?
or add this line `     "trim_trailing_white_space_on_save": false` in the "preference" > "settings - user".
Maybe it would be help.

closed 2016-01-28 by gepd

aMoniker, 2016-01-28:
Mine are set to Unix. The issue is unresolved and should not have been closed.

gepd, 2016-01-28:
Robot-will will not update this branch anymore, instead he will release a new version of it, see https://github.com/Robot-Will/Stino/issues/285