Robot-Will/Stino #232: Grayed out menus

Duality4Y, 2015-01-23:
Hi there!
so far Stino amazing!
I had some trouble after i installed packagecontrole,
and got grayed out menu like this:
https://github.com/Robot-Will/Stino/issues/194

what happend was that package controle (i think)
made a backup folder put the stino folder in a folder in there.
in my case it was under:
~/.config/sublime-text-3/Backup/20150123183731

I fixed it by copying the stino folder back to:
~/.config/sublime-text-3/Packages

and then it worked again ! once i closed the editor and reopend it didn't work again.
the folder was removed from the Packages folder.. 

this sometimes happens and sometimes doesn't.

Duality4Y, 2015-01-23:
creating a simlink to the place of the stino folder renamed the symlink to:
Stino-new-stino.package-control-old

gonzoid, 2015-01-23:
Could you try solution in #221

Duality4Y, 2015-01-26:
jeej so far it seems to work !!

closed 2015-01-26 by Duality4Y