Robot-Will/Stino #233: serial port changes after each upload to arduino Leonardo board

digitmovie, 2015-01-25:
Hello,
I am using  Sublime text 3 with Stino plugin. 
After I upload to arduino Leonardo the code , the serial port switches automatically to 1. for example I have the arduino installed on serial port: "com 3" and after  the upload of the code is finished to arduino, the setting it switches back to "com 1". 
If I want to upload a new program, I have to go back in the menu and set again the serial port to "com 3" every time. And only after that I can upload again. During development process of a software it is necessary to upload the code to arduino many times, so this is very annoying and time consuming.
How can I set up the plugin to stay on the right port and to not change every time I upload a new code?

Thank you.

juanjodcv, 2015-01-25:
hello

arduino yun has same problem for me, in MAC.
another serial problem that i don’t know if is related, in serial monitor,
if you activate “send” in serial monitor menu to send something to arduino,
after press enter to send, you must to enable send command again each time you need
to send one command to arduino, this is unusefull, most time you need to send more than one command.
It will be better to be “always enabled send”, as arduino IDE.

thank you Will, awesome job, thank you very much
juanjo - spain

fernandofranca, 2015-02-20:
Same issue with arduino nano.
Thanks!

dulouie, 2015-03-03:
hi guys,
i have a simple fix for this issue.
change in serial_port.py in line 113 the
        target_serial_port = serial_ports[3]
for COM4 as a example.

closed 2017-03-11 by Robot-Will