Robot-Will/Stino #234: new install of Stino as package in new install of Sublime Text 3 in Windows 8.1 machine fails

jmkpost, 2015-01-26:
I did a first-time installation of Sublime Text 3 on my computer in Windows 8.1, then attempted to install Stino-new-stino as follows:

Downloaded the Stino-new-stino branch of Stino as a zip file
Clicked the menu Preferences->Browse Packages in Sublime Text
Extracted the stino zip file contents into the Sublime Text Packages folder
Restarted Sublime Text.

There is no Arduino menu added in Sublime Text.

I first tried using the 64 bit Sublime Text program, and then uninstalled that and repeated the process with the 64 bit portable version of Sublime Text. Both times after restarting Sublime Text, there is no menu addition for Stino/Arduino.

pthieu, 2015-01-28:
On OSX Yosemite, menu gets created and placed into Packages/User/Stino, but Packages/Stino-new-stino is deleted. Not sure if this is a stino thing or an ST3 thing.

gonzoid, 2015-01-28:
Please try the solution given at #221

pthieu, 2015-01-30:
Thanks. It works now but getting another error:

"/Users/pthieu/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avrdude" "-C/Users/pthieu/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/etc/avrdude.conf" -v -v -v -v -pattiny85 -c{upload.protocol} "-P/dev/tty.wchusbserial1410" -b{upload.speed} -D "-Uflash:w:/Users/pthieu/Documents/Arduino/bin/Blink/Blink.hex:i"

avrdude: invalid baud rate specified '{upload.speed}'

My configuration is Programmer: ArduinoISP as ISP; Board: ATtiny85 (Internal 1MHz Clock).

I can successfully program Arduino with ArduinoISP sketch, but issues arises when trying to program ATtiny.

closed 2017-03-09 by Robot-Will