Robot-Will/Stino #237: The menu options that are not deployable, are disabled.

Herokid, 2015-02-13:
First of all, thank you infinitely that you shared the work you've done with Stino.

My problem is this:
I've been reading some issues that spoke of disabled menus, but did not solve my problem.
Although the title of the issue is quite descriptive, attached two screenshots that best illustrate the problem.

[image]

[image]

Work with:
Mac OS 10.10.2
Arduino 1.6.0
ST3 Build 3065

I honestly do not know how to get more information about the error.

Thank you very much in advance.
David.

closed 2015-02-15 by Herokid

alexfomenko, 2015-02-19:
I have the same issue! How did you managed with it?

Herokid, 2015-02-19:
Yo need remove package-metadata.json 
<https://github.com/Robot-Will/Stino/blob/new-stino/package-metadata.json> 
file from unzipped Stino folder before installation.

David.

gonzoid, 2015-04-28:
This is a duplicate of #221