Robot-Will/Stino #242: Need to single quote -D flags

yingyuho, 2015-02-27:
Line 46 of Stino/preset/platform_sam.txt:
build.usb_flags=-DUSB_VID={build.vid} -DUSB_PID={build.pid} -DUSBCON -DUSB_MANUFACTURER={build.usb_manufacturer} -DUSB_PRODUCT={build.usb_product}

Change this:
-DUSB_PRODUCT={build.usb_product}
To:
'-DUSB_PRODUCT={build.usb_product}'

Otherwise shell will eat double quotes in build.usb_product ("Arduino Due" in my case)

Osyrus, 2015-03-30:
This fixed the first error that brought me here, however now I get a new error (c'est la vie);

"arm-none-eabi-g++: error: '-DUSB_PRODUCT=Arduino Due': No such file or directory"

jurnaza, 2015-03-31:
I get the same error on Windows, but the above fix does work for OS X.

AndrewCapon, 2015-04-01:
The above fix doesn't work for me on OS X.

judepereira, 2015-07-19:
Thanks! This worked for me (single quotes)

@AndrewCapon while editing that file, make you aren't in the rich mode of TextEdit on your Mac. It will automatically cause that special one to show up.

closed 2017-03-09 by Robot-Will