Robot-Will/Stino #246: Menus disabled when installing into Sublime3

scottdky, 2015-03-06:
Removed older Stino and manually installed per readme. Package gets moved to User packages, but all the menus are disabled.

mannkind, 2015-03-21:
I was having the same issue - this work-around (mentioned by another GitHub user somewhere; I don't take credit for) worked for me.

Before copying the Stino-new-stino folder into the Sublime Text Packages folder, remove the file `Stino-new-stino\package-metadata.json` file.

davidbeermann, 2015-03-27:
Thanks @mannkind. Worked for me. I assume that the problem is actually caused by the *Package Manager* package – hence the name of the file, package-metadata.json. The manually installed package *Stino-new-stino* will be moved to the *Backup* folder once the Package Manager detects the 'illegal' or 'improper disabled' Stino package. The *Stino* folder inside of the User folder was always created, since it contains the menu structure of the Arduino menu. The menu items are disabled since it can't find the actual package. If you look into the file *'Stino-new-stino/stino/pyarduino/pyarduino.settings'* you'll find that Stino relies on two paths: *package_path* and *user_path*. And it can't find the package path, once the folder gets moved to the Backup folder. That's why the menu items are disabled. At least this is my conclusion after spending the last hour trying to install the new Stino.

MrLeeh, 2015-04-20:
Just move the *stino* folder back from *Backup* to *Packages* and delete *package-metadata.json*.

aniston, 2015-04-28:
"Before copying the Stino-new-stino folder into the Sublime Text Packages folder, remove the file Stino-new-stino\package-metadata.json file."
-- the above worked for me too.

gonzoid, 2015-04-28:
Try the solution given in #221. If this fixes your problem, please close your ticket :)

roddersg, 2015-05-12:
This works for me too on Ardiuno-1.6.4.

closed 2015-05-24 by scottdky

scottdky, 2015-05-24:
This worked for me too. I liked the explanation given by davidbeermann. After the fixing this problem, the avr-g++ compiler could not be found. I applied the fix described in https://github.com/Robot-Will/Stino/issues/89, only to have it prepend extra path info and still not working. Finally, after backing out the patch, it found the path just fine - strange. It almost seems like stuff gets cached, and takes some prodding to get it flush out. Thanks for all the help. I can finally use it now, and I really like it so far.