Robot-Will/Stino #248: /bin/sh: arm-none-eabi-g++: command not found

cosmikwolf, 2015-03-17:
Stino does not work with the latest version of Teensyduino (1.21)

I believe it is because there is a new flag in the board.txt file which Stino does not read properly.

teensy31.build.toolchain=hardware/tools/arm/bin/

This flag specifies where the arm-none-eabi-g++ compiler lives relative to the Arduino 1.6.1 Java folder.

Steps to reproduce:
Install Sublime Text
Install Latest Stino
Install Arduino 1.6.1
Install Teensyduino 1.21 on to Arduino 1.6.1
https://www.pjrc.com/teensy/td_download.html
Attempt to compile any sketch.

cosmikwolf, 2015-03-31:
this issue still exists with Teensyduino 1.22b and Arduino 1.6.2

luxcapto, 2015-06-01:
Issue confirmed on OS X.

brainchen98, 2015-07-05:
issue exists with teensyduino 1.24 on windows 7

brainchen98, 2015-07-05:
is there a fix to this?

HanYangZhao, 2015-07-10:
same problem here

yekohat, 2015-08-12:
Same problem with Teensyduino 1.24 and Arduino 1.0.6

yekohat, 2015-08-12:
Ok I understood how to fix this, you have to install Package Control who is a plugin for Sublime Text (2 & 3). Then follow the tutorial on the index page of this repo (Don't forget to unistall previous version of Stino).

cosmikwolf, 2015-08-27:
I found a fix. the problem is in the stino platform file. 

1. install Teensyduino
2. install Arduino 
3. Replace the contents of 

    > ~/Library/Application Support/Sublime Text 3/Packages/Stino/preset/platform_teensy3.txt

  with the contents of 

    > /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/platform.txt
4. Compile in stino!

cosmikwolf, 2015-08-29:
I also had to modify /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/boards.txt 

change:

teensy31.build.flags.ld=-Wl,--gc-sections,--relax,--defsym=__rtc_localtime={extra.time.local} "-T{build.core.path}/mk20dx256.ld"

to:

teensy31.build.flags.ld=-Wl,--gc-sections,--relax "-T{build.core.path}/mk20dx256.ld"

luxcapto, 2015-08-29:
Thanks cosmikwolf!!! It worked on my Mac. For anyone else encountering this, make the changes and restart Sublime Text as not all changes are updated when Sublime is running. On a side note, my Arduino Java folder was put inside the Resources folder when you open up the Arduino app package. i.e. /Applications/Arduino.app/Contents/Resources/Java/....

luxcapto, 2015-08-29:
Works on Linux (Arch) too :) Directory structure is a bit different though. Arduino will be/ in /usr/share/arduino. Sublime text config should be in ~/.config/sublime-text-3

cosmikwolf, 2015-12-01:
Issue still exists wtih Arduino 1.6.6, and Teensyduino 1.26, although now I got the same error. I resolved it by changing the platform_teensy3.txt file in the Stino preset folder. I changed one of the first few lines to:

compiler.path={runtime.ide.path}/hardware/tools/

a ghetto rig, but it works!

jebuschrast, 2016-07-01:
Is the addition of:
compiler.path={runtime.ide.path}/hardware/tools/
go to the same  platform_teensy3.txt that has the contents of platform.txt?

parkerlreed, 2016-12-14:
No luck here. Tried the edit but it still throws the same error.

Robot-Will, 2017-03-11:
I have passed Teensy build with the current version.

closed 2017-03-11 by Robot-Will