Robot-Will/Stino #252: Teensy Libs break Stino compile paths

bo01ean, 2015-03-27:
I installed Teensy on top of Arduino IDO and it has broken Stino :(
I love this Stino thing you created.

Here is the compile/verify log:

[Stino - Start building "src"...]
[  2%] Creating /tmp/Stino_build/src/src/Bridge.cpp.o...
In file included from /Applications/arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/Arduino.h:223:0,
                 from /Applications/arduino.app/Contents/Resources/Java/hardware/teensy/avr/libraries/Bridge/src/Bridge.h:22,
                 from /Applications/arduino.app/Contents/Resources/Java/hardware/teensy/avr/libraries/Bridge/src/Bridge.cpp:19:
/Applications/arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.h: In member function 'HardwareSerial& HardwareSerial::operator=(const HardwareSerial&)':
/Applications/arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.h:83:7: error: non-static const member 'volatile uint8_t* const HardwareSerial::_ubrrh', can't use default assignment operator
 class HardwareSerial : public Stream
       ^
/Applications/arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.h:83:7: error: non-static const member 'volatile uint8_t* const HardwareSerial::_ubrrl', can't use default assignment operator
/Applications/arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.h:83:7: error: non-static const member 'volatile uint8_t* const HardwareSerial::_ucsra', can't use default assignment operator
/Applications/arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.h:83:7: error: non-static const member 'volatile uint8_t* const HardwareSerial::_ucsrb', can't use default assignment operator
/Applications/arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.h:83:7: error: non-static const member 'volatile uint8_t* const HardwareSerial::_ucsrc', can't use default assignment operator
/Applications/arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.h:83:7: error: non-static const member 'volatile uint8_t* const HardwareSerial::_udr', can't use default assignment operator
In file included from /Applications/arduino.app/Contents/Resources/Java/hardware/teensy/avr/libraries/Bridge/src/Bridge.cpp:19:0:
/Applications/arduino.app/Contents/Resources/Java/hardware/teensy/avr/libraries/Bridge/src/Bridge.h: In member function 'void SerialBridgeClass::begin(HardwareSerial&, long unsigned int)':
/Applications/arduino.app/Contents/Resources/Java/hardware/teensy/avr/libraries/Bridge/src/Bridge.h:111:14: note: synthesized method 'HardwareSerial& HardwareSerial::operator=(const HardwareSerial&)' first required here 
       serial = _serial;
              ^

[Stino - Exit with error code 1.]

PaulStoffregen, 2015-03-27:
Which version of Arduino is this?  When installing Teensyduino, did install some, all or none of the optional extra libraries?

These small details, not present in your message, matter greatly for resolving problems like this.

bo01ean, 2015-03-27:
[image]

I think I installed all of them:

[image]

The sketch is using the Yun Bridge lib, but I think the Teensy one is getting pulled instead. Anyway to omit the the Teensy Bridge one?

PaulStoffregen, 2015-03-27:
Turn on verbose info while compiling in File > Preferences.

Then after Verify or Upload, scroll up to the top of the message.  It'll tell you exactly which libraries it actually used.

closed 2017-03-11 by Robot-Will