Robot-Will/Stino #254: Arduino 1.6.2 compatibility

sjoerdtimmer, 2015-03-29:
At least on archlinux I'm having issues with stino and the recent release of arduino 1.6.2.
(for instance my /usr/share/arduino/hardware folder is now empty). At first it complained about a missing avr-g++ (which should reside in /usr/share/arduino/hardware). I decided to reinstall Stino. Installation has never been smooth(it complains on the first run with a vague error) but this time I cannot get it to recognise the arduino application directory. It sais okay but ask me to set it when I try to compile...

sjoerdtimmer, 2015-03-29:
Partly fixed it: It turns out I had a version of stino from subl Package control interfering with my uninstall. 
I can now verify that stino works with arduino 1.6.1.

sjoerdtimmer, 2015-03-29:
With 1.6.2 I'm still experiencing isseus: all options in the Arduino>Preferences submenu are grayed out.

magic890, 2015-03-30:
+1, same problem here.

Osyrus, 2015-03-31:
It seems the Arduino compiling system is radically different. I see the preferences fine, but no boards show up. (On Windows 8.1). The boards not showing up makes sense because the hardware directory no longer has anything in it except "listComPorts". The tools folder in the Arduino root now contains a Mangler thingo.

Robot-Will, 2015-03-31:
This plugin do not support 1.6.2 currently. With Arduino 1.6.2 a whole new tools and library packaging is introduced. The contents of the hardware directory is now found as a compressed archive in the new dist directory. The avr tools are installed by the Arduino IDE to the directory ~/.arduino15 (on Linux) or ~/Application Data/Arduino15 (on Windows).

davidpn, 2015-04-01:
I'm a first time user of Arduino so I have 1.6.2, and discovered Stino. 

I'm having the same issues as the others, however I managed to have Stino (on OSX) populate the Boards menu by extracting default.package.tar.gz, and moving the avr folder to under /Contents/Java/hardware (and restart Sublime Text). 

Trouble is now that Stino fails to find avr-g++ when attempting to compile. I figured that moving the avr-gcc folder to under /Contents/Java/tools would solve the issue. Perhaps I need to put it somewhere else?

sjoerdtimmer, 2015-04-01:
If you're a new arduino user I strongly recommend using the arduino IDE. It may not have the best code editor, but to get started it is really easy. If you insist on using stino you can get older arduino versions from the arduino site: http://arduino.cc/en/Main/OldSoftwareReleases I recommend 1.6.0 for this purpose. Most changes in 1.6.2 relate to arm board anyway so you wont miss out on the important features.

davidpn, 2015-04-01:
I want to use Stino mainly for the code editor, yes. As good as the Arduino IDE is, the editor is very lacking by comparison. Thanks for the tip; I've installed 1.6.0 and I'm up and running.

tuhtah, 2015-06-25:
In case it helps someone else, I was able to patch up my setup using Sublime Text 2 on Arch Linux, Arduino 1.6.5 and the previous Stino code base ("version": "2014.02.25.05.15.29") installed from Package Control using the patch below.

The patch fixes both the greying out issue and not being able to find the compiler due to the new Arduino IDE package directory structure:

```diff
--- a/app/base.py
+++ b/app/base.py
@@ -838,7 +838,7 @@ def getKeywordListFromFile(keywords_file):
                                if len(word_list) == 3:
                                        keyword_type = word_list[1]
                                        keyword_ref = word_list[2]
-                               elif len(word_list) == 2:
+                               else:
                                        if 'LITERAL' in word_list[1] or 'KEYWORD' in word_list[1]:
                                                keyword_type = word_list[1]
                                                keyword_ref = ''
--- a/app/compiler.py
+++ b/app/compiler.py
@@ -313,6 +313,7 @@ def getDefaultArgs(cur_project, arduino_info):
 
        arduino_folder = base.getArduinoFolder()
        ide_path = os.path.join(arduino_folder, 'hardware')
+       avr_path = os.path.join(ide_path, 'tools', 'avr')
        project_name = cur_project.getName()
        serial_port = getSelectedSerialPort()
        archive_file = 'core.a'
@@ -322,6 +323,8 @@ def getDefaultArgs(cur_project, arduino_info):
 
        args = {}
        args['runtime.ide.path'] = arduino_folder
+       args['runtime.tools.avr-gcc.path'] = avr_path
+       args['runtime.tools.avrdude.path'] = avr_path
        args['ide.path'] = ide_path
        args['build.project_name'] = project_name
        args['serial.port.file'] = serial_port
```

ioanniskar, 2015-09-11:
I have downloaded Stino from Package Control. Where are these files I have to edit?

hypotamtam, 2015-09-12:
To find them, open Sublime then Preferences->Browse Packages. That should open an window. On this one, go in Arduino-like IDE\app.

hackermagnet, 2015-09-12:
Thanks I had installed a different version of Stino with different file structure.
I made these changes but I couldn't select the Arduino app folder.
I changed on base.py the line:
```folder = os.path.join(folder, 'Contents/Resources/Java')```
to
```folder = os.path.join(folder, 'Contents/Java')```
and I was able to select the Arduino app folder but still doesn't work.
If I hit verify/compile I get this message:
```Compiling rcxvf...
Creating /Users/username/Documents/Arduino_Build/rcxvf/rcxvf.ino.cpp.o...
/Users/username/Documents/Arduino_Build/rcxvf/rcxvf.ino.cpp:1:21: fatal error: Arduino.h: No such file or directory
 #include <Arduino.h>
                     ^
compilation terminated.
[Stino - Error 1]```

MacroMachines, 2016-01-31:
Is there any chance this might get fixed? I am working on a large project and it is very difficult trying to bounce between 3 different versions of arduino and different library versions to get back to a nice fluid workflow.  I really dislike working in arduino IDE now that I spent months working in sublime and fell in love wit it, but can not seem to get stino back up and running.  I am mainly using teensy 3.1 which I believe benefits from the new features, at least it appears my project runs smoother with the new libraries and 1.6.7 IDE.

gepd, 2016-01-31:
Have you downloaded the last version of the `new-stino` branch? 
Some days ago I've pushed a fix that solved the problem in most of the Arduino IDE (In my tests)

As Robot-Will will not update longer this stino version, I won't do it too, he said he will release a new version of the plugin, but isn't a release date yet.

I'm working in a new plugin called [Deviot](https://github.com/gepd/Deviot) it's not using the arduino IDE instead use [platformIO](http://platformio.org/) as compiler, it currently is in beta, if you like to try it and maybe report any error/bug or request new feature, you will be welcome.

MacroMachines, 2016-02-04:
downloaded the new-stino and replaced my current version, tried arduino 1.6.7 and I get this error:
[  1%] Creating /Users/AxiomCrux/Desktop/000/Sketchbook/_Builds/OmniMod_1.30/OmniMod_1.30.ino.cpp.o...
/bin/sh: arm/bin/arm-none-eabi-g++: No such file or directory

works fine in the actual IDE

gepd, 2016-02-04:
Did you follow [this](https://github.com/Robot-Will/Stino#installation) steps to install it?
I'm currently running the `new-stino` with the version `1.6.7`, what board are you trying to compiling?

dwhacks, 2016-02-14:
I was able to get the new-stino to work but not with my boards installed via the board manager in Arduino.
Any way for this to work? I would like to use it with the ESP8266

Duality4Y, 2016-02-14:
i install the esp board not through the board manager but through github.
the read me has a guide on how to do it:
https://github.com/esp8266/Arduino
under using git version.

danfitz7, 2016-10-21:
@MacroMachines , did you ever get Teensy + Sublime + Stino working together?

MacroMachines, 2016-11-02:
I had it working great a year ago, and then it stopped working when I updated my arduino IDE.  I now use CLI arduino with Sublime sometimes, and sometimes just use the arduino IDE to upload with the "Use External Editor" enabled in the preferences.  CLI is kinda slow cause it launches the whole arduino IDE but immediately quits after (like its just doing the compile and upload and quits)

stino might work, its been a long time since I tried it.  I notice I had missed a message from this thread.

parkerlreed, 2016-12-14:
Same issue with 1.6.13. The base.py file doesn't exist anymore so I'm not sure what to edit.

Using the Stino from https://github.com/gepd/Stino/tree/new-stino via the Package Installer

closed 2017-03-09 by Robot-Will

Robot-Will, 2017-03-10:
@parkerlreed  Currently, I am working on a totally new plugin (https://github.com/Robot-Will/Stino) to make all things work.

parkerlreed, 2017-03-10:
Ahh thanks!