Robot-Will/Stino #256: Can't select Arduino's folder

powermanwxy, 2015-04-04:
Why there is NO respond when I choose Arduino 's folder on quick panel? My system is OS X 10.10.3,and my Arduino IDE is 1.6.0.

markusk, 2015-04-21:
Same here. 

Sublime Text 3, Stable Channel, Build 3083
Arduino 1.6.3
Mac OS X 10.10.2

powermanwxy, 2015-04-22:
I am already solved this problem,your Arduino 's version is too high,try to change to1.2.X

markusk, 2015-04-22:
Version 1.2.X is *centuries old* and not available for download: http://www.arduino.cc/en/Main/OldSoftwareReleases#previous

I found the path correctly set in file **stino.settings**:

```python
{
    "arduino_folder": "/Applications/Arduino.app", 
    "baudrate": 4, 
    "board": 0, 
    "full_compilation": true, 
    "platform": 1, 
    "platform_name": "Arduino AVR Boards", 
    "serial_port": 5
}
```

But it is still asking for me to set it... Is there an option within the package to make it "believe", that the path is correctly set? Any manual change in any other file or code?

Any help appreciated.

emj365, 2015-04-23:
Same here my IDE is 1.6.3.

```
{
    "arduino_folder": "/opt/homebrew-cask/Caskroom/arduino/1.6.3/Arduino.app",
    "full_compilation": true,
    "line_ending": 3,
    "serial_port": 4,
    "verify_code": true
}
```

not works

topgenorth, 2015-05-02:
Having the exact same problem with Arduino 1.6.3.

ilighthouse, 2015-05-05:
I had the same problem, fixed by 
go to /Users/Name/Library/Application Support/Sublime Text 2/Packages/Arduino-like IDE/app/base.py and change line 284:
folder = os.path.join(folder, 'Contents/Resources/Java')
to
folder = os.path.join(folder, 'Contents/Java')

markusk, 2015-05-05:
Thanks.
Does *not* work with here (Sublime Text 3):

```python
Compiling abc...
Creating /Users/markus/Documents/Arduino_Build/abc/abc.ino.cpp.o...
/bin/sh: avr-g++: command not found
[Stino - Error 127]
```

imaginezz, 2015-07-12:
I have the same issue,
I use Arduino on my surface pro 3(Work with sublime text 3),but I can't select the arduino folder on my SD card or SSD.
My folder is D:\Programs\Arduino

mark2n, 2015-08-28:
Just in case that some Windows user come here looking for similar issue's solution, here is my experience:
1) In stall "Package Control". 
I firstly followed this link(http://en.code-bude.net/2015/05/29/tutorial-configuring-sublime-text-as-arduino-ide-with-stino/), it worked perfectly until the step to select Arduino's installation folder. 
2) In stall the package "Stino".
Then I just followed the installation instruction(https://github.com/Robot-Will/Stino). It worked like magic.

Good luck!

JoaoAlpinista, 2015-11-06:
I'm running OS X and had the same problem.
I solved it by using Arduino 1.6.0 instead of the latest.

jshardy, 2016-05-13:
I found a solutions lol. You aren't going to believe this... STOP typing in the arduino location folder!!! There is a text box, look below it. There is a / showing which folder it is in. Cick on the / and navigate to to each folder. When you finally get to the right one it will just disappear and work.... I can't believe something so stupid wasted hours of my time lol.

diimdeep, 2016-09-30:
@jshardy you're my hero.

Robot-Will, 2017-03-09:
Stino2017 does not need to select Arduino App Folder. Thanks a lot.

closed 2017-03-09 by Robot-Will