Robot-Will/Stino #258: Don´t upload sketch.

jalfonsosuarez, 2015-04-07:
I've recive this message when Start Uploading:

[Stino - Start uploading...]
"{runtime.tools.avrdude.path}/bin/avrdude" "-C{runtime.tools.avrdude.path}/etc/avrdude.conf" -v -patmega2560 -cwiring "-PCOM11" -b115200 -D "-Uflash:w:D:\programacion\arduino projects\build\Test_Tipos/Test_Tipos.hex:i"
El sistema no puede encontrar la ruta especificada.
[Stino - Exit with error code 1.]

In Arduino IDE 1.6.3 works fine.

Thanks.

khairulhasanmd, 2015-04-13:
I've got the same error in my Windows 7 Laptop.
[Stino - Start uploading...]
"{runtime.tools.avrdude.path}/bin/avrdude" "-C{runtime.tools.avrdude.path}/etc/avrdude.conf" -v -patmega328p -cusbasp -Pusb "-Uflash:w:D:\Arduino\Build\Arduino-Robot-Arm/Arduino-Robot-Arm.hex:i"
The system cannot find the path specified.

[Stino - Exit with error code 1.]

Don't know how to fix it :(

kswain1, 2015-04-18:
Received a similar error when compiling avrdude: stk500_recv(): programmer is not responding. Can't download code unto the board.

MrLeeh, 2015-04-20:
The same on my Windows 7 Laptop:

```
[Stino - Start uploading...]
"{runtime.tools.avrdude.path}/bin/avrdude" "-C{runtime.tools.avrdude.path}/etc/avrdude.conf" -v -patmega328p -carduino "-PCOM36" -b115200 -D "-Uflash:w:C:\Users\Lehmann\AppData\Local\Temp\Stino_build\ustempctrl/ustempctrl.hex:i"
[Stino - Exit with error code 1.]
```
I guess the part in the brackets should be replaced by the actual path.

Sporakamis, 2015-05-01:
I'm on 8.1 and have the same error.

"{runtime.tools.avrdude.path}/bin/avrdude" "-C{runtime.tools.avrdude.path}/etc/avrdude.conf" -v -patmega2560 -cwiring "-PCOM3" -b115200 -D "-Uflash:w:C:\Users\Judson\AppData\Local\Temp\Stino_build\Blink/Blink.hex:i"
The system cannot find the path specified.
[Stino - Exit with error code 1.]

I've tried editing paths, but I don't know how to fix this. I would really appreciate some help if anyone else knows how to handle this.

khairulhasanmd, 2015-05-01:
Hi,
I don't know how to fix it but i have a previous working backup in my HDD. 
If you need, you can download from here:
https://drive.google.com/file/d/0B4zy7eqONHnuajk2aXlzOTJtaDQ/view?usp=sharing

miaoudak, 2015-06-03:
Have the same problem v1.6.4

kdschlosser, 2015-06-09:
here is the fix

search for 
params['software'] = 'ARDUINO'
under the pyarduino folder
in the file 
arduino_target_params.py

you will see a list of params. at the bottom add
params['runtime.tools.avrdude.path'] = '/whatever/you/have/for/the/parent/folder/for/avrdude'

note***** it adds the /etc/avrdude.conf and the /bin/avrdude

tis why i said to add the parent folder

tonyputi, 2015-06-12:
You need only to add this at line 323 of your stino/pyarduino/arduino_target_params.py
```python
target_tools_avrdude_path = os.path.join(ide_path, 'hardware/tools/avr')
```
and this at 332 of same file

```python
params['runtime.tools.avrdude.path'] = target_tools_avrdude_path
```

kdschlosser, 2015-06-16:
How about windows systems. ? I am not that familiar with python. But this is one hell of an upgrade from the standard arduino ide. And using visual studio I would love to. But still new to programming. And it's alittle overwhelming and I do like the fact that this loads fast. I really don't think that for an  arduino using a full feature ide like visual is necessary.

fabio978, 2015-06-25:
tonyputi's solution worked for me on Windows 8.1! Thanks a lot, i can now use my favorite editor also for Arduino!

nono40, 2015-11-10:
hi
I have the same problem with win 8.1 and arduino 1.6
Uploading...
"/bin/avrdude" "-C/etc/avrdude.conf" -v -patmega2560 -cwiring -PCOM29 -b115200 -D "-Uflash:w:C:\Users\a\Documents\Arduino_Build\test_transmission/test_transmission.hex:i"
Le chemin d'accŠs sp‚cifi‚ est introuvable.
[Stino - Error 1]

In the kdschlosser' solution of the 9 june where are pyarduino folder and the file arduino_target_params.py. Impossible to find them
Thank you for your help

closed 2016-01-28 by gepd

nono40, 2016-01-28:
hello
where are pyarduino folder and the file arduino_target_params.py in win8.1
I can't do the kdschlosser' solution if I don't find them
thank you very much

gepd, 2016-01-28:
From the ST menu go to `Preferences > Browse Packages...` look for the `Stino` folder.
The `arduino_target_params.py` file is in `Stino\stino\pyarduino`

nono40, 2016-01-29:
hank you  very much for your answer

I installed ST2 and ST3 It is the same thing: impossible to upload  the *.hex on the arduino board

I looked in preferences/browse packages and there is not stino folder

I make a screen copy of packages browser and the folders in the packages are very different beetween ST2 and  ST3

You can see them in attached files

Thank you very much for helping me

[image]
[image]
[image]

gepd, 2016-01-29:
As you have an old version of stino, the folder of the plugin is called `Arduino-like-IDE` (in the last image) but you have also installed `Stino-master` maybe could be that too. You have to try with one of those folders.

nono40, 2016-01-29:
thank you very much It 's good now

I explain for the other people
I remove arduino-like-ide and stino-master like this:
preferences/package control/remove package and select  package to remove
after I install the last stino package as le chapter installation here: https://github.com/Robot-Will/Stino and add this :" params['runtime.tools.avrdude.path'] = 'hardware\tools\avr\bin': " in arduino_target_params.py file like explained by kdschlosser on 9 Jun 2015 (see above)
I opened  .py file with python

Thank you very much gepd for helping me

nono40, 2016-01-30:
hello
I was too happy . It was good only once
I have always an error:
[Stino - Téléversement en cours...]
"hardware	oolsBELvrBSin/bin/avrdude" "-Chardware	oolsBELvrBSin/etc/avrdude.conf" -v -patmega2560 -cwiring "-PCOM29" -b115200 -D "-Uflash:w:C:\temp\test/test.hex:i"
La syntaxe du nom de fichier, de répertoire ou de volume est incorrecte.

I translate: The syntax of the file name, directory or volume is incorrect.
[Stino - Terminé avec le code d'erreur 1.]

The file" test.hex" is created but always  impossible to upload to the arduino board
I don't know where to see in order to find the problem

thank you for helping me

gepd, 2016-01-30:
Are you using the last version of the branch `new-stino` ? 
There is not much I can do, as thoses branches will not longer updated because Robot-Will will release a new version of the plugin (not release date yet)

I'm working a new plugin called [Deviot](https://github.com/gepd/Deviot), if you like you can try it, it's currently in beta, but you should't have problem to upload an sketch, it's supporting more than 200 boards, you can help me repporting any bug or new feature you want.

Turban-dominique, 2016-05-28:
bonjour,

version arduino 1.6.9
subblime text ok et acheté
stino installé ok.

je n'arrive pas a téléverser > voir en dernière ligne.
le port est bon:
la carte est bonne:
cela compile bien:

bien cordialement
Dominique.

[Stino - Compilation de "afficheur_barre_8x7segts" en cours...]
[ 25%] Création de E:\Documents and Settings\ELECTRONIQUE\ARDUINO_compilation\afficheur_barre_8x7segts\afficheur_barre_8x7segts.ino.cpp.o...
[ 50%] Création de E:\Documents and Settings\ELECTRONIQUE\ARDUINO_compilation\afficheur_barre_8x7segts\afficheur_barre_8x7segts.elf...
[ 75%] Création de E:\Documents and Settings\ELECTRONIQUE\ARDUINO_compilation\afficheur_barre_8x7segts\afficheur_barre_8x7segts.eep...
[100%] Création de E:\Documents and Settings\ELECTRONIQUE\ARDUINO_compilation\afficheur_barre_8x7segts\afficheur_barre_8x7segts.hex...
Ce projet utilise 1,558 octets (4.8%) de mémoire sur les 32,256 octets disponibles.
Les variables globales utilisent 14 octets (0.7%) de mémoire dynamique, laissant 2,034 octets pour les variables locales. Le total disponible étant de 2,048 octets.
[Stino - Compilation de "afficheur_barre_8x7segts" terminée en 1.1s.]
[Stino - Téléversement en cours...]
Le chemin d'accŠs sp‚cifi‚ est introuvable.

[Stino - Terminé avec le code d'erreur 1.]