Robot-Will/Stino #259: /bin/sh: 1: /bin/avrdude: not found

eni23, 2015-04-11:
when trying to upload it fails because stino tries to access avrdude via:

"/bin/avrdude" "-C/etc/avrdude.conf" -v -patmega32u4 -cusbasp -Pusb -e -Ulock:w:0x3F:m -Uefuse:w:0xcb:m -Uhfuse:w:0xd8:m -Ulfuse:w:0xff:m
/bin/sh: 1: /bin/avrdude: not found
[Error 127]

i tried to grep in the code for that but not found anything related to that. workaround is symlinks but this is really a ugly solution

twxia, 2015-05-15:
same as here~
my avrdude is located in /usr/bin/ not in /bin/avrdude

eni23, 2015-05-16:
the issue is fixed in 1.6.4, see https://github.com/arduino/toolchain-avr/issues/23 for more details..

closed 2015-05-16 by eni23

twxia, 2015-06-01:
hi @eni23 I'm using 1.6.4 but it still exist~

terxw, 2015-06-04:
Same as @twxia, I am using 1.6.4, but the problem still exists

cimenx, 2015-06-10:
Same as @terxw and @twxia , i'm using arduino 1.6.4 linux 64bit but avrdude still not located
cc: @eni23

shimanyi, 2015-08-28:
@twxia @terxw @eni23 got any work-arounds on this issue?

shimanyi, 2015-08-28:
@Robot-Will Hi, any solutions on this issue?

twxia, 2015-08-28:
@shimanyi u can have a soft link to solved it  (`ln -s ....`)

shimanyi, 2015-08-28:
@twxia I am a bit new to this :( How do I do that? Thanks

shimanyi, 2015-08-28:
So guys, I found a fix, I think it is just some configurations I had wrong. Nonetheless, I downgraded a little to 1.6.0, previously running on 1.6.3 and it worked like charm.