Robot-Will/Stino #264: compile issue: '"avr-g++"' is not recognized as an internal or external command,  operable program or batch file.

mnbf9rca, 2015-04-20:
I've installed the latest version (.zip) to ST2, disabling the older version, although i had this issue in that too. I've selected the arudino folder (C:\Program Files (x86)\Arduino), containing Arudino 1.6.3. When i click Arduino > Verify/Compile I get the error below. I can see g++.exe at C:\Program Files (x86)\Arduino\hardware\tools\avr\avr\bin

    [Stino - Start building "v1"...]
    [  3%] Creating C:\Users\R\AppData\Local\Temp\Stino_build\v1\v1.ino.cpp.o...
    '"avr-g++"' is not recognized as an internal or external command,operable program or batch file.
    [Stino - Exit with error code 1.]

Duodugs, 2015-04-21:
You have to add "Arduino 1.6.3 folder"\hardware\tools\avr\bin to your path.

dmsherazi, 2015-04-27:
can you please explain how I can do that?

I am facing the same problem and I have already added the arduino program path and the sketchbook folder path, I can see all boards and libraries in the menu.

mnbf9rca, 2015-04-27:
thanks @Duodugs.
@dmsherazi first google hit is e.g. http://stackoverflow.com/questions/9546324/adding-directory-to-path-environment-variable-in-windows :)

closed 2015-04-27 by mnbf9rca