Robot-Will/Stino #267: No Board Found

TimBak, 2015-04-23:
Hello,

I'm having a "No Board Exists" issue when trying to compile.

Windows 8
Sublime Text 3.
Downloaded the zip and placed it in the appropriate folder.
Arduino menu appears, as expected!
Set the Application Folder, get "Arduino 1.6.2 is found at C:\Program Files (x86)\Arduino."
Set the Sketchbook Folder, now the sketches appear in Arduino -> Sketches

However, no boards show up in Arduino -> Board or Arduino -> Board Options
And, of course, when I try compiling, get the following error:
No board exists. Please Select Arduino Application Folder or Change Arduino Sketchbook Folder in Arduino Menu -> Preferences.

Any suggestions?
I've searched here and elsewhere to no avail.
Thanks

JayPalm, 2015-05-01:
I have the same problem on my rMBP using Arduino 1.6.2 and ST build 3083. It simply says "none" when I scroll over Board or Board options.
[image]

TimBak, 2015-05-05:
OK, looks like there is two of us!

Jay,
Where did you find out about the 1.6.2 compatibility issue?

jakkubu, 2015-06-16:
I had similar problem. I didn't have any examples, libraries too. In my case I hadn't set proper Arduino Application Folder.
In Sublime go to Arduino >> Preferences >> Select Arduino Application Folder 
In my case it was located in "C:\Program Files (x86)\Arduino" 
Do you have 64 bit Windows? Maybe it's because of different install folder name?

runevn, 2015-07-04:
I'm having the same issue. 

The Arduino Application folder is set but nothing in Board or Board Option. 

I'm running Arduino 1.6.2 and OS X 10.10.4.

Sevenskey, 2016-07-18:
I have the same problem. 
I'm running Arduino 1.6.9 and Arch with Linux kernel 4.6.4.

Dominicguri, 2016-10-31:
I having the same problem. Did anyone solve this problem yet?

Robot-Will, 2017-03-11:
Please try current version, this version does not depend on Arduino IDE.

closed 2017-03-11 by Robot-Will