Robot-Will/Stino #271: /bin/sh: avr-g++: command not found Mac OS X 10.10.3

sebasqure, 2015-05-01:
Every time I try to verify or upload the code, e get "/bin/sh: avr-g++: command not found error code 127".

Arduino IDE compiles with no problem but Stino is much better...

yamikiri, 2015-05-03:
I have similar problem, but its on Win8.1.
Fixed by modifying a file under Arduino folder.
You may check $(Arduino-installed-folder)/hardware/arduino/avr/platform.txt
Comment original line: # compiler.path={runtime.tools.avr-gcc.path}/bin/
Add extra line under it: compiler.path={runtime.ide.path}/hardware/tools/avr/bin/

Side effect not confirm yet....

ilighthouse, 2015-05-05:
Same problem here. New installation of Arduino (1.6.4). I am getting:
/bin/sh: avr-g++: command not found
[Stino - Error 127]

sebasqure, 2015-05-05:
I discover the solution. Give 10 min an I publish it.

sebasqure, 2015-05-05:
After download the folder "Stino-new-stino" from here: https://github.com/Robot-Will/Stino/wiki/Manual-installation on step 2, unzip it and access "..../Stino-new-stino/stino/pyarduino/arduino_compiler.py" open it in a code editor and comment the original function "exec_cmd" and replace it by:

def exec_cmd(working_dir, cmd):
    os.environ['CYGWIN'] = 'nodosfilewarning'
    if cmd:
        os.chdir("/")
        cmd = formatCommand(cmd)
    if "avr-" in cmd:
        cmd = "Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/" + cmd;

        cmd = cmd.replace("{runtime.tools.avrdude.path}", "Applications/Arduino.app/Contents/Java/hardware/tools/avr/")

        compile_proc = subprocess.Popen(cmd, stdout=subprocess.PIPE,
                                    stderr=subprocess.PIPE, shell=True)
        result = compile_proc.communicate()
        return_code = compile_proc.returncode
        stdout = result[0].decode(base.sys_info.get_sys_encoding())
        stderr = result[1].decode(base.sys_info.get_sys_encoding())

    else:    
        return_code = 0
        stdout = ''
        stderr = ''
    return (return_code, stdout, stderr)

be carful with the indentation it should be something like the attached image.

PS: the arduino app should be on main application folder and I'm supposing you are on a mac.
![captura de ecra 2015-05-5 as 07 58 26]

(https://cloud.githubusercontent.com/assets/12198921/7468130/8c764ffa-f2fc-11e4-9738-d61d806492be.png)
 
Hope this help.

sebasqure, 2015-05-05:
The program becomes buggy if you install "Package control" maybe you can download "Sublime text 2" and use it only with this package. You souhld install the stino package manually.   I did this and it works like a charm.

xvoland, 2015-05-10:
@sebasqure  Thanks! It's works for me (OSX 10.10.3, Arduino 1.6.4)

Alnoa, 2015-05-11:
same here , had to upload with regular arduino ide

baduncadonk, 2015-05-19:
I'm on OSX 10.8.5, Sublime Text 2, and Arduino 1.6.3.  I played around with the manual-installation instructions and found the attached change to arduino_compiler.py to work well, but only in tandem with changing the Preferences under the new Arduino menu (at the top right of ST2) to point "Select Arduino Folder" to the Arduino app, in top level of Applications folder, and pointing "Change Sketch Folder" to the folder holding my sketches.

I couldn't find the Package Control command, but that doesn't seem to have prevented ST2 from working well with a variety of Arduino boards.  If someone has insights on where to find Package Control, I'd love to take a look at it.  I'm also not confident that I know where the Arduino build files reside. 

[image]

gepd, 2015-06-08:
Everyone who want to collaborate, can do it in this fork: https://github.com/gepd/Stino

objectiveSee, 2015-07-14:
I was able to get it working using sebasqure's solution! :100:

TMcGahee, 2015-07-24:
@yamikiri Thanks so much! Worked like a charm, Win8.1

davidtaboas, 2015-07-29:
A best solution for the code of @sebasqure https://github.com/Robot-Will/Stino/issues/271#issuecomment-98975026 is indent the lines that concerns the "avr-" commands. With this modifications upload still working fine in osx 10.10.3 (in my case):
```python
def exec_cmd(working_dir, cmd):
    os.environ['CYGWIN'] = 'nodosfilewarning'
    if cmd:
        os.chdir("/")
        cmd = formatCommand(cmd)
        if "avr-" in cmd:
            cmd = "Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/" + cmd;
            cmd = cmd.replace("{runtime.tools.avrdude.path}", "Applications/Arduino.app/Contents/Java/hardware/tools/avr/")

        compile_proc = subprocess.Popen(cmd, stdout=subprocess.PIPE,
                                    stderr=subprocess.PIPE, shell=True)
        result = compile_proc.communicate()
        return_code = compile_proc.returncode
        stdout = result[0].decode(base.sys_info.get_sys_encoding())
        stderr = result[1].decode(base.sys_info.get_sys_encoding())

    else:    
        return_code = 0
        stdout = ''
        stderr = ''
    return (return_code, stdout, stderr)
```

TakeMatsu, 2015-07-30:
@sebasqure's solution worked like a charm!! Now I stino can compile and upload, thank you!!

closed 2015-08-03 by gepd