Robot-Will/Stino #274: Uploading problem on ST 2 Yosemite 10.0.3

sebasqure, 2015-05-05:
@Robot-Will can't you please help me? This plugging is very good but I'm having some problems.

- To select arduino application folder I had to change a function from your code as I mentions on issue #271, by now app folder is selected;

- The plugin only works if "Package control" is not installed, that's why I'm using ST2... (only for arduino);

- Now the problem the is most important cause the other I manage to work around them. When compelling, it says that the code has been sent but the arduino don't do nothing. (If I use arduino app with the same code and USB port it works)

Thanks

andreucv, 2015-05-09:
@Robot-Will 
Same issue and same problem.

When uploading it compiles the code perfectly. Then it says: "Start uploading..." and "Done uploading." but Arduino (Nano, in my case in /dev/tty.usbserial-A603XXXX) don't flash any LED for TX and RX. 

According to you ( @sebasqure  ) if I use same code and USB port and programmer in Arduino.app it upload the code to the Board!

sebasqure, 2015-05-09:
@andreucv 
I managed to make a work around on this. Star by downloading Arduino 1.0.5 from Arduino.cc website and place it in a folder in Applications folder. Then delete and reinstall the stino but when reinstalling don´t modify any code in the files. Then when selecting the application folder in Sublime, go to the folder you created earlier and select arduino 1.0.5 instead of 1.6.4 this way it should work. At leas for me it worked.

mediaartdesign, 2015-05-27:
@sebasqure 
Finally! That did it for me! Thanks a lot!!!

closed 2017-03-09 by Robot-Will