Robot-Will/Stino #275: avr-g++: command not found when compiling, using Arduino IDE 1.6.2

simonliu009, 2015-05-08:
While using Arduino IDE 1.6.2 and Sublime Text3, there is a error message says "avr-g++: command not found when compiling", and then it stops.

And I roll back to Arduino IDE 1.0.5 and it is working fine.

sebasqure, 2015-05-09:
That's happening cause Arduino 1.6.2 and above have some files that stino needs in a different locations... see my issue #271, for instructions on how to solve that erro. Although after solving the at least for me it uploads the code but nothing appends to the arduino. So I had to go back to 1.0.5 . And I think that there is one more thing, package control is causing some trouble to stino.

otrees, 2015-05-27:
'"avr-g++"' 不是内部或外部命令，也不是可运行的程序或批处理文件 。Arduino IDE 1.6.4
"avr-g++: command not found" . using Arduino IDE 1.6.2

MrLeeh, 2015-05-27:
Same for me... Using Windows 7.

gepd, 2015-06-08:
everyone who want to collaborate can do it in this fork: https://github.com/gepd/Stino

gepd, 2015-06-10:
This issue was fixed in the new fork

ad-on-is, 2015-06-14:
The issue still exists for me.
SublimeText 3
Arduino IDE 1.6.4

MrLeeh, 2015-06-14:
I tried the new fork but still got the issue.

gepd, 2015-06-15:
I had add a new information for a momentary solution, it should fix it https://github.com/gepd/Stino

please, new issues report it in the previous fork

Super169, 2015-07-07:
I just try the following command in Console
  setx path "%path%;E:\Arduino\IDE\arduino-1.6.4"
where E:\Arduino\IDE\arduino-1.6.4 is the installation folder of Arduino IDE.

But the following error is returned:

>>> setx path "%path%;E:\Arduino\IDE\arduino-1.6.4"
  File "<string>", line 1
    setx path "%path%;E:\Arduino\IDE\arduino-1.6.4"
            ^
SyntaxError: invalid syntax

Is this command for the Console of Sublime Text 3?

gepd, 2015-07-07:
Super169 that is not a command for SB, it's for windows console

zazurich, 2015-07-08:
FIXED:

Hi eveyone. I am using windows 8 and Arduino IDE v1.6.5

For windows, just try the following command in windows Console
setx path "%path%;C:\Program Files (x86)\Arduino\hardware\tools\avr\bin"

Super169, 2015-07-13:
Thanks all, the problem is solved. It has to update the path in Environment Variables.

testard, 2015-07-23:
I am having the same results as sebasqure (as I understand it) after implementing the solution from #271: Sublime Text/stino reports compiling/verifying and uploading the code, but my Arduino board does not receive it. I tried both ST 2 and ST 3 without Package Control installed and I'm running Arduino 1.6.5 on OS 10.9.5.

Is there a Mac (Unix) version of the Windows console "setx path..." fix?

globmont, 2015-07-28:
Try following this tutorial http://www.cyberciti.biz/faq/appleosx-bash-unix-change-set-path-environment-variable/ and adding the path of your installation plus the path to the bin folder similar to what Super169 had.

davidtaboas, 2015-07-29:
In OS X 10.10.3: https://github.com/Robot-Will/Stino/issues/271

gepd, 2015-07-30:
I've add some fixes to the current branch, please download and test the plugin.

closed 2015-08-02 by gepd