Robot-Will/Stino #278: Stino-new-stino folder is deleted after reboot SB3

gepd, 2015-05-26:
I'm trying to install the plugin in a fresh sublime text 3 installation.

But every time I copy the `Stino-new-stino` folder and I restart SB3, it delete `Stino-new-stino` from the package folder.

I tried install only the plugin and also installing the old Stino, and after disabling it, in both case the result is the same. Is this happening to anyone else?

alberto2000, 2015-05-26:
Yes, exact same problem here.

gepd, 2015-05-27:
Meanwhile I've solved this issue deleting the package control, it's not he optimal solution, but at leats worked for me.

alberto2000, 2015-05-28:
That's no solution for me, as I depend on the Package Manager quite a lot. Any other hint on this?

alberto2000, 2015-05-28:
Found a potential solution and made a pull request - https://github.com/Robot-Will/Stino/pull/280

gepd, 2015-05-28:
none of this fixes solve that problem I tryed all of them.

For my is a solution because I can keep working with stino, and can install manually the rest of the plugins.

jggdotcom, 2015-06-06:
For any package control users still having this problem... Instead of installing manually by dropping in the downloaded package folder, you can add this repository to package control via the command palette (Package Control: Add Repository). Then, run the 'Install Package' command and choose the package named 'Stino', NOT the package in the default repository called 'Arduino-Like IDE'.

After installing in this way everything is working for me. 

Hope that helps someone.

ikarnauk, 2015-06-08:
Thanks jggdotcom, worked for me.

gepd, 2015-06-09:
jggdotcom It work for me too I had updated the instalation instructions in this new fork: https://github.com/gepd/Stino

closed 2015-06-26 by gepd