Robot-Will/Stino #281: Migrate to new owner?

varmint09, 2015-06-01:
The owner does not seem to be maintaining this library, and there are many outstanding PR, is there anyone else who would be willing to fork this and maintain it?

As it stands its too unstable for me to use with ST3 on OSX 10.10... and that seems to be commonplace.

gepd, 2015-06-08:
I forked the project to help in the updates, until now the new installation process wrote by **jggdotcom** in this issue page https://github.com/Robot-Will/Stino/issues/278 seems to work

sdaitzman, 2015-06-17:
I concur with this issue. I think moving forward on gepd's fork unless @Robot-Will would like to resume development would be a sensible step to take, and I'm willing to help out there.

gepd, 2015-06-18:
Last connection of Robo-Will was on March. My idea is work and improve the issues of the plugin to keep working with it, and if he want to resume the development, I can send a pull request, so he can start in the actual state.

forwardadvance, 2015-06-26:
I got an email from Robot Will today. His gmail account was blocked by the Chinese government, so he's been out of touch. He has added me as a collaborator. I have merged a few of the simple PRs but my free time is limited.

gepd, 2015-06-26:
I'll to help in all I can, by now just have to wait for the plans of rebot in the new pugin.

I'll close this "issue" so we can comment in the new announcement of robot-will https://github.com/Robot-Will/Stino/issues/285

closed 2015-06-26 by gepd