Robot-Will/Stino #285: Would you like to join into this software development？

Robot-Will, 2015-06-26:
First of all, thanks for all of your supports.

Arduino IDE is not a finished software and the software and directory structure of the IDE are changing and changing. After its change, the plugin should be rewrote to support the new version of Arduino IDE. But it is hard to keep compatibility to the new and old version IDEs. Moreover, it is messy of third-party cores and libraries. It resulted problems of lacking standard directory structures.

So I will not update the current plugin and I will write a new software. The new plugin will not follow Arduino IDE directory structure and it will be better. The following days I will give information about the new software.

 I created a new branch "ST4ArduinoIDE" of the current plugin based on the branch "new-stino"，and if you want to make it better, please leave message here and I will add you into collaborators. Thanks.

spambus, 2015-07-17:
TFTP support would be nice for uploading sketches to arduino's with the Ariadne Bootloader...

az-z, 2015-10-13:
Robot, thank you for your work. don't know nearly enough to contribute the code.

closed 2017-03-11 by Robot-Will