Robot-Will/Stino #287: Stino does not recognise Arduino MIDI library

matthearn, 2015-07-19:
Hi, 

I am working on an hardware MIDI device using the Arduino MIDI Library https://github.com/FortySevenEffects/arduino_midi_library/ and unfortunately Stino doesn't seem to see it. It works fine in the Arduino IDE But not Sublime Text. Could you give me any pointers as to where I am going wrong.

Cheers,

Matt H

gepd, 2015-07-22:
Can you give us, the version of the IDE you're using, S.O and Sublime text. It can help to start.

matthearn, 2015-07-22:
I am using arduino 1.6.5, Sublime Text Build 3083, and as for the version of Stino, I cannot tell you as the help dialogue does not seem to be working, but I downloaded the Github Zip package last week so that should give you some idea. I am running this all on 64bit Ubuntu Mate 14.04.2 LTS. Sorry, I would have provided all of this in my initial bug report but I was obviously suffering some form of mental block, oops!

Cheers,

Matt H

matthearn, 2015-08-20:
Can anyone help me please?

Eternyt, 2016-06-16:
see #350

closed 2017-03-09 by Robot-Will