Robot-Will/Stino #288: No option for "Arduino Due" board

chchmatt, 2015-07-21:
Am I missing something here? Googled it and searched issues here, but don't see anyone with the same problem. Under Arduino -> Boards there's no "Arduino Due" board to select... hence my uploads don't work.

gepd, 2015-08-02:
Arduino Due is currenty not supported, I'll try to add that board next days, it can took a little bit, because I don't have the board, so I'll need people like you to test it

Leamucho, 2015-08-03:
I'll be testing too

t3chguy, 2015-08-06:
:+1: I'll be able to test also.

eliasse, 2015-09-07:
Me too :+1:

yuvalsegev11, 2015-10-18:
Is there any update about the Due support?

rafaellehmkuhl, 2015-10-19:
I would like too :+1:

Grippentech, 2015-11-22:
Waiting for that as well. Really hoping to get it soon, after becomming so used to Stino its annoying to get back to having to use the IDE.

closed 2017-03-09 by Robot-Will