Robot-Will/Stino #289: compile succeeds, Stino says upload succeeds, but does not actually upload

hwestbrook, 2015-07-22:
On a mac, 10.10.4, Stino succesfully compiles my sketches, then says it uploads them, but doesnt actually upload. Taking the command from the console and pasting into terminal succesfully uploads the sketch to device. Same behavior for both usb upload and upload with AVR ISP MKII. 

USB:
````
[Stino - Done building "application" in 0.0s.]
[Stino - Start uploading...]
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude" "-C/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf" -v -patmega328p -carduino "-P/dev/cu.Bluetooth-Incoming-Port" -b115200 -D "-Uflash:w:/tmp/Stino_build/westbrook-labs-endpoint-application/westbrook-labs-endpoint-application.hex:i" -V
[Stino - Done uploading.]
````

AVR ISP MKII:
````
[Stino - Done building "application" in 0.0s.]
[Stino - Start uploading...]
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude" "-C/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf" -v -patmega328p -cstk500v2 -Pusb "-Uflash:w:/tmp/Stino_build/westbrook-labs-endpoint-application/westbrook-labs-endpoint-application.hex:i" -V
[Stino - Done uploading.]
````

Taking the commands noted in the console and pasting them into terminal succesfully uploads the sketch.

davidtaboas, 2015-07-29:
I resolved a similar problem with this: https://github.com/Robot-Will/Stino/issues/271#issuecomment-125946097

gepd, 2015-07-29:
I've been merge a fix base on https://github.com/Robot-Will/Stino/issues/271 please download and test it.

closed 2015-08-02 by gepd