Robot-Will/Stino #290: Switch to @PlatformIO cross platform code builder

ivankravets, 2015-07-23:
Hey friends,

I'm an author of [PlatformIO](http://platformio.org). What do you think about replacing build logic which is based on Arduino Application to @PlatformIO code builder and give users ability to use more then 150+ different embedded boards and 10+ development platforms? 

Existing solution http://docs.platformio.org/en/latest/ide/sublimetext.html

gepd, 2015-07-29:
Interesting solution **ivankravets**, I'll read about it, maybe it can be agreat solution to solve much of the problem this plugin have.

ivankravets, 2015-07-29:
Thanks @gepd. I'm glad to help you with the PlatformIO API (JSON response):
```shell
# available boards
platformio boards --json-output

# available platforms
platformio platforms search --json-output

# installed platforms
platformio platforms list --json-output

# available serial ports list
platformio serialports list --json-output

```

gepd, 2015-12-01:
Hi @ivankravets I've been looking in your application, implementing it into this plugin would be a really big improvement.

I ran the examples in the windows console, and the sublime text.
As I never implemented a ST plugin from the beginning, maybe you can help me with something, I can't run the `platformio init --ide sublimetext --board %TYPE%` command

I've already implemented a function to create the ini file, but it getting hard to run this without the build system, if you have some info, or any site with this info I'll appreciate.

gepd, 2015-12-01:
never mind, I already figured out.

ivankravets, 2015-12-01:
Hi @gepd,

Don't hesitate contact me for any questions :+1:

gepd, 2015-12-01:
Thanks, I'll need for sure.

gepd, 2015-12-04:
@ivankravets There is way to do `platformio init -d directory` auto accepting the creation of the folders and ini file?
I'm using `subprocess.Popen` to run this command and it can't finish because I don't know how to answer yes to the prompt `Do you want to continue? [y/N]: `

ivankravets, 2015-12-04:
@gepd sure! Please take a look on the current template https://github.com/platformio/platformio/blob/develop/platformio/ide/tpls/sublimetext/platformio.sublime-project.tpl

Please run all commands in this format
```
platformio -f -c sublimetext CMD
```

where CMD could be `init`, `run`, etc.

See docs for `-f/--force` http://docs.platformio.org/en/latest/userguide/index.html

ivankravets, 2015-12-04:
Also, I've removed "auto-uploading" from `init` command. See https://github.com/platformio/platformio/commit/5dc1396f05ba03e0b697e7e606427bf2f6ece4c7

Before PlatformIO <2.4.2 you can use `platformio -f -c sublimetext init --disable_auto_uploading -d directory` and from >=2.4.2 (has not been released yet) please use `platformio -f -c sublimetext init -d directory`

I hope that you will propose users 4 commands:
1. Build `platformio -f -c sublimetext run -d directory` 
2. Build & Upload `platformio -f -c sublimetext run -t upload -d directory` 
3. Clean `platformio -f -c sublimetext run -t clean -d directory` 

Don't forget that `-d/--project-dir` isn't required. You can change `CWD` to project directory and run command from it.

gepd, 2015-12-04:
Thanks for that update. I was just looking for this options.
Now look very simple to continue with the implementation.

I'm using the explicit way `-d directory` cause the way how the stino was developed.

Your system will bring a lot of improvement to this plugin.

ivankravets, 2015-12-04:
How I can try this snapshot? I would like to see how does it work. Thanks!

gepd, 2015-12-04:
I'll upload a first version ASAP. I'll let you now.
I'll like to talk you about the new features I'm thinking to add with the help of your platform, including the library manager, I think will be very good to not depend of the Arduino IDE at all.

ivankravets, 2015-12-04:
Please feel free to [contact me](https://github.com/ivankravets) if you have any questions or comments.

gepd, 2015-12-04:
I started to do some test with the build process and I received this error

```
C:\Users\user\Desktop\blink_example>platformio run
[12/04/15 18:21:14] Processing uno (platform: atmelavr, targets: upload, framewo
rk: arduino, board: uno)
-------------------------------------------------------------------------------
avr-gcc -o .pioenvs\uno\firmware.elf -Os -mmcu=atmega328p -Wl,--gc-sections,--re
lax -L.pioenvs\uno -Wl,--start-group -lm .pioenvs\uno\libFrameworkArduinoVariant
.a .pioenvs\uno\libFrameworkArduino.a -Wl,--end-group
.pioenvs\uno\libFrameworkArduino.a(main.o): In function `main':
C:\Users\user\Desktop\blink_example/.pioenvs\uno\FrameworkArduino/main.cpp:37:
undefined reference to `setup'
C:\Users\user\Desktop\blink_example/.pioenvs\uno\FrameworkArduino/main.cpp:47:
undefined reference to `loop'
collect2.exe: error: ld returned 1 exit status
scons: *** [.pioenvs\uno\firmware.elf] Error 1
========================= [ ERROR ] Took 1.58 seconds =========================
```
This is from the windows console, some days ago when I tested it with the same example (took from the platformio web) by the console I had not problem.

ivankravets, 2015-12-04:
See compiler errors `undefined reference to `setup'`... Do you have any files in `src` directory? Look like your project is empty.

gepd, 2015-12-04:
you're right, I thought when you initialize the project, it was automatically copied, my fault.

gepd, 2015-12-05:
This is a fast snapshot https://github.com/gepd/Stino/archive/StinoIO.zip

At this moment I only tested with Arduino UNO, I still have a lot of changef to do in the code

gepd, 2015-12-05:
I forgot to mention, at this moment, the upload option is the only available.

closed 2015-12-14 by gepd

ivankravets, 2015-12-14:
The further development is moved to https://github.com/gepd/Deviot