Robot-Will/Stino #293: serial monitor not work

patomoi, 2015-07-31:
hi, i'm using sublime text 3, and the last version of stino, and i try use the serial monitor, and he dont work, obviously i check the correct port (in this case COM5), and i uploaded to arduino UNO a sketch with Serial.begin(9600)  and check  in stino menu the correct baud rate,  finally, i check all and all is fine

sorry my bad english, greets from chile

gepd, 2015-08-02:
what is your S.O?

patomoi, 2015-08-02:
hi, thanks for your attention, my OS is windows 7 ultimate.

marcoskubis, 2016-03-27:
Any progress about this? Same problem here, i'm not able to open some serial monitor.

gepd, 2016-03-27:
There is not progress in this issue, the plugin is not longer updated. Please read this: https://github.com/Robot-Will/Stino/issues/331

closed 2017-03-22 by Robot-Will