Robot-Will/Stino #295: Sublime 3 erasing stino ST4 every reboot

Alnoa, 2015-08-03:
and i cant do anything :/

gepd, 2015-08-03:
Are you installing it manually or by the package control? (https://github.com/Robot-Will/Stino#installation)

Alnoa, 2015-08-03:
manually  yeah ST4

Alnoa, 2015-08-03:
ok dont know why bu isnt erasing anymore , but cant compile now , returning a directory or syntax error .
[image]
it say " syntax file name , repository or volume is incorrect"

gepd, 2015-08-03:
I'm sorry, I meant ST3. 

To avoid ST delete the folder you have to follow the installation instructions: https://github.com/Robot-Will/Stino#installation

To try to fix that bug, please:

mark "Show compilation output" in the menu Arduino-> Build Options
mark "Show upload output" in the menu Arduino-> Build Options

try to compile again, when it shows the error copy it and paste here, after that go the console `Menu >View > Show console` and copy the output too.

Finally tell me you O.S.

Alnoa, 2015-08-03:
[image]

DPI scale: 1
startup, version: 3083 windows x64 channel: stable
executable: /C/Program Files/Sublime Text 3/sublime_text.exe
working dir: /C/Program Files/Sublime Text 3
packages path: /C/Users/Alex/AppData/Roaming/Sublime Text 3/Packages
state path: /C/Users/Alex/AppData/Roaming/Sublime Text 3/Local
zip path: /C/Program Files/Sublime Text 3/Packages
zip path: /C/Users/Alex/AppData/Roaming/Sublime Text 3/Installed Packages
ignored_packages: ["Vintage"]
pre session restore time: 0.0759392
warning: mnemonic TE not found in menu caption Text
startup time: 0.102939
first paint time: 0.102939
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_file_settings
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.quick_panel
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin 0_package_control_loader.00-package_control
reloading plugin 0_package_control_loader.02-bz2
reloading plugin Alignment.Alignment
reloading plugin BracketHighlighter.bh_core
reloading plugin BracketHighlighter.bh_logging
reloading plugin BracketHighlighter.bh_plugin
reloading plugin BracketHighlighter.bh_regions
reloading plugin BracketHighlighter.bh_remove
reloading plugin BracketHighlighter.bh_rules
reloading plugin BracketHighlighter.bh_search
reloading plugin BracketHighlighter.bh_swapping
reloading plugin BracketHighlighter.bh_wrapping
reloading plugin DocBlockr.jsdocs
reloading plugin DuplicateSelections.duplicate_selections
reloading plugin Generate Password.password
reloading plugin GenerateUUID.generate_uuid
reloading plugin Inc-Dec-Value.inc_dec_value
reloading plugin Package Control.bootstrap
reloading plugin Package Control.Package Control
reloading plugin Select Quoted.select_quotes
reloading plugin Table Editor.table_base
reloading plugin Table Editor.table_border_syntax
reloading plugin Table Editor.table_emacs_org_mode_syntax
reloading plugin Table Editor.table_lib
reloading plugin Table Editor.table_lib_test
reloading plugin Table Editor.table_line_parser
reloading plugin Table Editor.table_multi_markdown_syntax
reloading plugin Table Editor.table_pandoc_syntax
reloading plugin Table Editor.table_plugin
reloading plugin Table Editor.table_plugin_test
reloading plugin Table Editor.table_re_structured_text_syntax
reloading plugin Table Editor.table_simple_syntax
reloading plugin Table Editor.table_textile_syntax
reloading plugin Table Editor.widechar_support
reloading plugin Trimmer.Trimmer
reloading plugin Wrap Plus.wrap_plus
reloading plugin Zen Tabs.TabsWorker
reloading plugin Zen Tabs.ZenTabs
reloading plugin Zen Tabs.ZenTabsTest
reloading plugin Stino.StinoStarter
warning: mnemonic TE not found in menu caption Text
warning: mnemonic TE not found in menu caption Text
reloading plugin SublimeLinter.commands
reloading plugin SublimeLinter.sublimelinter
plugins loaded
SublimeLinter: debug mode: off 
Limit: 10
Full path: False
Highlight: True
warning: mnemonic TE not found in menu caption Text
Package Control: Skipping automatic upgrade, last run at 2015-08-04 00:19:04, next run at 2015-08-04 01:19:04 or after
reloading Packages/User/Preferences.sublime-settings
warning: mnemonic TE not found in menu caption Text
generating syntax summary
warning: mnemonic TE not found in menu caption Text
warning: mnemonic TE not found in menu caption Text
warning: mnemonic TE not found in menu caption Text
warning: mnemonic TE not found in menu caption Text
warning: mnemonic TE not found in menu caption Text
warning: mnemonic TE not found in menu caption Text
warning: mnemonic TE not found in menu caption Text
warning: mnemonic TE not found in menu caption Text
warning: mnemonic TE not found in menu caption Text

OS win8.1

gepd, 2015-08-03:
it said is problem with file name or the volume is incorrect.

Just to test, can you change the name of your file (like test.ino), and change it to a new location, like in the main hard drive (usually C:/)

Alnoa, 2015-08-04:
problem stay :/ with previous stino version no problems

Alnoa, 2015-08-05:
Arduino upgraded to 1.6.5 resolve the issue (was 1.6.3)

closed 2015-08-05 by Alnoa