Robot-Will/Stino #296: Stino don't work with 1.6.1/1.6.0

Kristey, 2015-08-05:
When installing I cannot choose the Arduino folder. I click on "Select current folder" and I can't select.
I also experimented with version 1.6.0, also doesn' t work

gepd, 2015-08-05:
do you instaled it manually or by the package control?

Alnoa, 2015-08-05:
need to clean your stino install and upgrade your arduino ide to 1.6.5

Kristey, 2015-08-05:
I installed for package control, don't worked.
After this I installed manually and don't worked again.

In this moment works with arduino 1.0.6

PureBasic, 2015-09-22:
Hello all,
Kind of the same problem here, I installed Arduino 1.6.5, I can select the folder but I get 
> [Stino - Start building "AnalogReadSerial"...]
> [  3%] Creating C:\Users\roehrb\AppData\Local\Temp\Stino_build\AnalogReadSerial\AnalogReadSerial.ino.cpp.o...
> avr-g++: error: '-DUSB_MANUFACTURER=Unknown': No such file or directory

> avr-g++: error: '-DUSB_PRODUCT=Arduino Leonardo': No such file or directory

> [Stino - Exit with error code 1.]

when trying to compile.

az-z, 2015-10-13:
I have it working against 1.6.5
at least i can compile the examples.

julianadenauer, 2015-10-14:
I have the same problem like @Kristey with Arduino 1.6.5 on Mac. Installed and removed Stino through package control several times. Didn't help.
Used to work like a charm for the last year. Really missing it now!

dobrMAN, 2015-10-23:
Do not work with Arduino 1.6.5 (try in Ubuntu Linux 15.04 and Windows 7) and ST2

andig, 2015-11-07:
Dooes not work with Arduino 1.6.5 on Windows 7 with ST3. Cannot select arduino folder.

Jim4Beam, 2015-11-12:
Does NOT work on Ubuntu 14.04 with ST2 nor ST3 and Arduino 1.6.5 (after choosing application folder, restart, all grayed out).
1.0.6 works.

gepd, 2016-01-24:
I've pushed a fix for this problem. In my tests it worked with Arduino IDE 1.6.0 - 1.6.5 - 1.6.7
Branch [New Stino](https://github.com/Robot-Will/Stino/tree/new-stino)

closed 2016-01-24 by gepd