Robot-Will/Stino #297: Boards only listing 'Arduino AVR Boards'

Spokkam, 2015-08-07:
Hey, I currently have the new-stino branch installed in Sublime and it's running fine. I can verify & upload to my Uno and it works. 

However, I wanted to switch to the new-stino branch because it supported the Intel Galileo board but I don't see it listed under Arduino > board.

I have Arduino 1.6.5-r3 installed. I'm using Sublime 3 & installed new-stino by zip folder. I'm pointing to the right application & sketches folder too. 

My Preferences.stino-settings are as follows: 

{
    "arduino_ide_path": "/Applications/Arduino.app", 
    "build_path": "/Users/gkelleher/Documents/Arduino", 
    "lang_id": "en", 
    "serial_port": "/dev/tty.usbmodemfd121", 
    "sketchbook_path": "/Users/gkelleher/Documents/Arduino", 
    "target_board_id": "ide.arduino.avr.uno", 
    "target_programmer_id": "ide.arduino.avr.avrisp"
}

Any help would be appreciated.

SteveDoesStuff, 2016-01-23:
I believe I'm having a similar issue. I've used the Arduino IDE to install addition boards (ESP8266 modules) but they don't appear in Stino's list of boards in Sublime

Surbyte, 2016-05-25:
Same issue. 
ST doesn´t recognize DUE board.
IDE 1.6.5 r3

willbelr, 2016-09-08:
I have a similar problem. I cannot see Digispark nor Attiny boards, altought it works fine in Arduino IDE.
I use Arch Linux with Arduino 1.6.11, Sublime text 2.0.2

closed 2017-03-09 by Robot-Will