Robot-Will/Stino #299: Doubled Arduino Application Path in build command

thinkyhead, 2015-08-16:
I'm using _Stino_ with _Sublime Text 2_ on _OS X 10.10.5_.

I used the __Select Arduino Application Folder__ command to select one of my Arduino IDE folders (I have a few). I then used the __Verify/Compile__ command, and got the following error:
```shell
/bin/sh: /Applications/_Applications/DIY/Arduino 1.6.1/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin//Applications/_Applications/DIY/Arduino 1.6.1/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-g++: No such file or directory
```
Note that the full path (`/Applications/...bin/`) is repeated before `avr-g++` in the command passed to `bin/sh`. I could probably make a PR for the issue if the Python code is easy to grok. I just need to be pointed in the right direction where the bug most likely is.

I also noticed that _Stino_ doesn't like _Arduino 1.6.2_ yet. Apparently the build tools for _1.6.2_ are installed in the user library folder. On my system the `avr-gcc++` tool is installed in `~/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino2/bin/`.

thinkyhead, 2015-08-16:
So far the _New Stino_ package is getting the application path correct. Still doesn't seem to know what to do with _Arduino 1.6.2_.

mloose, 2015-10-15:
I have the same problem, but with 1.6.5. 
Did you find a solution? Right now I'm trying to determine which file contains the incorrect path so I can correct it. If I find it, will let you know.

robin7331, 2015-10-31:
I found a way to fix this issue in some way. This is a pretty quick solution and I am not sure what other bugs might be caused with my change. But for now I am able to compile again. Should be fine till this is officially fixed.
I simply inserted `self.params['compiler.path'] = ''` at line 277 right after the if statement in file `{Sublime Text 3 Packages}/Stino/stino/pyarduino/arduino_compiler.py`

closed 2017-03-09 by Robot-Will