Robot-Will/Stino #301: ST 2 and ST3 Windows error on preparing core source files

hunnybear, 2015-08-18:
When I try to compile in both sublime text 2 and 3, I get the following error:

[  2%] Creating C:\Users\Tyler\AppData\Local\Temp\Stino_build\porto_pointer\porto_pointer.ino.cpp.o...
avr-g++: error: '-DUSB_MANUFACTURER=Unknown': No such file or directory
avr-g++: error: '-DUSB_PRODUCT=Arduino Micro': No such file or directory

[Stino - Exit with error code 1.]

Is this related to the single quotes being not necessary/bad? 

I tried doing the setx thing I saw in another issue, as well as adding them to my environment path, 
This works fine on my linux box, but I need to get this working on my windows machine in the next week or so.

PureBasic, 2015-09-22:
Same probleme for me. Did you find a solution ?

ckaminski, 2015-10-02:
I had the same problem - I tried uploading before compiling, and quite possibly before I finished unzipping my Arduino install.  

Which Arduino version? 

I reunzipped my Arduino install, then compiled and then uploaded and zero issues.

MarikSx, 2015-11-08:
W7, X:\Program Files (x86)\Arduino\hardware\arduino\avr, chande name or del file "platform.txt", restart ST2 or ST3. Enjoy!

gepd, 2016-01-24:
I've pushed a fix for this problem. In my tests it worked with Arduino IDE 1.6.0 - 1.6.5 - 1.6.7
Branch [New Stino](https://github.com/Robot-Will/Stino/tree/new-stino)

closed 2016-01-24 by gepd

sibinsivalal, 2016-12-08:
I believe you need to set the **board preference !**
under Menu>Arduino>Board>"_which ever board you are on_" for me its Arduino/Genuino UNO.

[image]