Robot-Will/Stino #303: the compilation panel is too small and always reset the size

hariseldon78, 2015-09-07:
hello, i would like to be able to see more of the compilation output in the panel, but the panel always appear at a small size and if i make it bigger the next time it still is small. Is there the possibility to change  this default size to something bigger, or better to something customizable? 
If this issue is of too low priority for your roadmap could you give me just a hint to which files control the panel size so i can fork and prepare a pull request?

thomasgrantferguson, 2015-10-01:
I agree!  It would be nice if either the window size was customizable, or would at least stay the same size after dragging it larger.  If you don't want to change it, could you at least point us in the direction where we can make the change ourselves?  I love Stino, and this is my only issue with using the program!  Thank you.

Robot-Will, 2017-03-11:
The previous version of this software opens a new penel each build process start. I do not know how to set the panels' size, and each time the panel appears, it is a new panel with the default size.

Stino2017 is a totally new software, there is only one message panel, and you can reopen it. So the panel size will keep the size.

closed 2017-03-11 by Robot-Will

thomasgrantferguson, 2017-03-15:
Thank you for all of your great work.  I love Stino!!!