Robot-Will/Stino #305: Singles quote mark around USB_FLAG mess up the compilation ST3+Arduino1.6.5r5

PureBasic, 2015-09-22:
Hi all,
Stino don't format the USB_FLAG arguments right when trying to compile for an USB tagret (leonardo for exemple). Here are the fail arguments:
> '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"'

Here are the arguments from the ArduinoIDE:

> -DUSB_MANUFACTURER="Unknown" -DUSB_PRODUCT="Arduino Leonardo"

I try to find a way to get rid of the quotation but I have just be turning around board.txt all day long without success :/

PureBasic, 2015-09-24:
Okay, I find a workaround by modifying the platform.txt file. Remove the quote mark at the end of the file. This is breaking the Arduino IDE so I made a copy of the whole arduino folder to be used exclusively by Stino.
I don't close the issue.

rafaellehmkuhl, 2015-10-21:
Can you explain the workaround better? Which version of the Arduino IDE are you using? I tried the procedure but all I found were new errors.

PureBasic, 2015-10-22:
I think I got new errors too, and switched to an Arduino Mega plateform... I check that for you tomorrow.

Robot-Will, 2017-03-09:
For each compile command, you can replace the command by the following codes. And Stino2017 has the codes.

```
pattern_text = r"'-D[\S\s]*?'"
pattern = re.compile(pattern_text)
texts = pattern.findall(cmd)
for text in texts:
     new_text = text[1:-1]
     cmd = cmd.replace(text, new_text)
```

closed 2017-03-09 by Robot-Will