Robot-Will/Stino #306: Teensy 3.1, teensy loader 1.25 compilation issues

gbernal, 2015-09-24:
I get this error when trying to compile for teensy 3.1
/Applications/Arduino.app/Contents/Java/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/bin/ld:--defsym:1: ignoring invalid character `{' in expression
/Applications/Arduino.app/Contents/Java/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/bin/ld:--defsym:1: ignoring invalid character `}' in expression
--defsym:1: undefined symbol `extra.time.local' referenced in expression
collect2: error: ld returned 1 exit status

[Stino - Exit with error code 1.]

wolfmanjm, 2015-10-03:
I soled this issue for 1.6.5 doing the following...

edit the file...
.../arduino-1.6.5-r5/hardware/teensy/avr/boards.txt 

replace the existing ```teensy31.build.flags.ld``` line with...

```teensy31.build.flags.ld=-Wl,--gc-sections,--relax,--defsym=__rtc_localtime=0 "-T{build.core.path}/mk20dx256.ld"```

Robot-Will, 2017-03-11:
I have passed build with the current version.

closed 2017-03-11 by Robot-Will