Robot-Will/Stino #312: Stino + Arduino Due

rafaellehmkuhl, 2015-10-21:
I think I have tried all the Stino versions with all the Arduino IDE versions and they all give some new error. Is there anybody successfully using Arduino Due with Stino today? If yes, whick combination do you use and how did you resolve the errors?

CatchMeFastFat, 2016-04-25:
I am using Arduino 1.6.0 + sublime 3 and working well with Arduino due(( I have tried serial))

Surbyte, 2016-05-25:
Arduino 1.6.5 + ST 3.0 can´t recognize DUE on board options.
Any help will be advised!

CatchMeFastFat, 2016-05-25:
As far as I know, you have to use older version in order to get it working.

new versions isn't supported yet(as I know).

So setup Arduino 1.6.0 and it will recognize it(you can get the same libs as the new one manually also compiler)

CatchMeFastFat, 2016-05-25:
Later versions could work ,but I am using 1.6.0"tried"

Surbyte, 2016-05-27:
Ok. Thanks.
I will discard STINO because DEVIOT runs well with new IDE versions.

closed 2017-03-09 by Robot-Will