Robot-Will/Stino #314: Stino - Not acessible Menus

guilhermegazzinelli, 2015-10-30:
Stino running in STX2. After the first openning and configuration, if you restart the menus get like inacessible. 

[image]

I tried to reinstall. Not acepting also the Package manager.

danielcrowley, 2015-11-20:
Any luck with this @guilhermegazzinelli? 
I'm hitting the same issue, seems to be after a restart after linking the arduino application folder.

jstnjns, 2015-12-30:
Same here.  Installed new Arduino recently, changed prefs to new location, everything disabled.

jwhendy, 2016-01-03:
I ran into a few issues recently, and posted a workaround for one of them on #322 . I *believe* I also had grayed out menus, but was able to get around them by installing Stino via Package Control instead of how I'd previously just copied the git folder to `~/Arduino/libraries` I just moved `~/.config/sublime-text-2` to `~/.config/sublime-text-2.bak` to see if I could recreate the process.

- reinstalled `sublime-text-2` via the Arch linux AUR
- start it up; no package control
- follow instructions for `st2` Package Control [here](https://packagecontrol.io/installation#st2)
- restart `st2` and get this pop-up

```
Package Control

Package Control just installed or upgraded the missing Python _ssl module for Linux since Sublime Text does not include it.

Please restart Sublime Text to make SSL available to all packages.
```

- use `preferences -> Package Control` to add `stino` repository and install `stino`
- restart `st2` for good measure
- get good menu

[image]

That said, I can't seem to recreate when I saw the empty menu! I tried copying the `stino` git folder to the `preferences -> Browse Packages` location (after wiping the whole `~/.config/sublime-text-2` folder again) but it shows/uploads. I tried reverting the change to `platform.txt` mentioned in #322 that caused build failures initially... still works! So, just adding the above in case it's at all helpful. I can't re-create the issue for myself.

guillaumedardenne, 2016-01-06:
Same issue ... Install package, change Arduino folder, reboot app, then no items available

guillaumedardenne, 2016-01-06:
Please could you be more precise about the process to get around this issue ?
I'm running on Mac os 10.11, sublime text 2 with a valid licence and Arduino 1.6.7

Thanks,

guillaumedardenne, 2016-01-07:
Got solved by reinstalling only... Works well now ! :)

jwhendy, 2016-01-07:
Could you elaborate? I definitely saw this happen initially but as stated could not reproduce. I think for future stumblers, it would be helpful to know the exact process. Your screenshot looks like Windows but you mentioned OS X... did you just do Control Panel -> Remove Applications. Delete the .app file? Delete ~/.config/sublime-text-x? Any other preference files? Just re-install `stino`? Things like that.

closed 2016-01-24 by gepd