Robot-Will/Stino #315: Error after upgraded Arduino IDE 1.6.6

Super169, 2015-11-09:
Just upgraded the Arduino IDE from 1.6.5r2 to 1.6.6, and it will return error of "No such file or directory"  for core.a whenever I compile the sketch, even for the dummy  blank sketch BareMinimum.
  
May I know if there has any setting need to be changed for using Arduino 1.6.6?

MarikSx, 2015-11-09:
Try next:
W7-W8-8.1, X:\Program Files (x86)\Arduino\hardware\arduino\avr, change name or del file "platform.txt", restart ST2 or ST3.
Enjoy!

fletchjeff, 2015-11-09:
I'm getting the following error on OS X:
`/Applications/arduino.app/Contents/Java/hardware/tools/avr/bin/avr-ar: {archive_file_path}: Permission denied`
Checked file permissions, its all good. The file runs fine from the command line.

MarikSx, 2015-11-09:
@fletchjeff try the find file with name "{archive_file_path}" in your system.
had the same problem on Win 7, in root disk was file "{archive_file_path}", i del him and problem disappeared.

fletchjeff, 2015-11-09:
I couldn't find a file like that on my systems. I did find are reference to it inside the platform.txt file though.

Could it be a version problem maybe? Stino thinks I'm running Arduino 1.6.6, but I'm running 1.6.9

bgiovand, 2015-11-10:
@fletchjeff I renamed platform.txt to platform.backup.txt and it worked. File is located here on OSX: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/

fletchjeff, 2015-11-10:
@bgiovand Tried that, but the problem remains. Any other options to try. I can't work out which file its trying to call.

fletchjeff, 2015-11-10:
I have dug a bit and in the platform.txt file there is a line:

`recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}"`
The compiler is failing when it gets to here, which I assume is specified by that line:

`"avr-ar" rcs  "{archive_file_path}" "/tmp/Stino_build/temp/core_arduino/hooks.c.o"
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-ar: {archive_file_path}: Permission denied`

Somewhere, something is not setting the `{archive_file_path}` value correctly. Does anyone know where that value should be set in the compile process? I can only find references to `archive_file_name`.

fletchjeff, 2015-11-10:
Right, seems to be fixed. I just removed all `platform*.txt` from the `/Applications/Arduino.app/Contents/Java/hardware/arduino/avr` directory, and restarted Sublime Text.

Super169, 2015-11-13:
Thanks a lot.
The problem can be fixed after I remove the platform.txt.
But will it affect the arduino IDE if I remove this file?

closed 2015-11-13 by Super169

fletchjeff, 2015-11-13:
@Super169 It does affect the IDE. I tried to open the IDE and it simply wouldn't load. I put the file back, and it loaded. But, much to my surprise, Stino still works. I have no idea why.

jonathansabbah, 2015-11-17:
Works for me !

MartinBloedorn, 2015-11-23:
@fletchjeff Had the same issue. Your fix (removing the file, reloading Stino, putting file back) works for me too. Thanks a lot!

halfnibble, 2015-11-29:
I just noticed that `arduino.app` is lower-cased in my error message, but the actual folder is with a capital "A" as in `Arduino.app`.

halfnibble, 2015-11-29:
It appears that @fletchjeff fix is a one-trick pony. Once you put the file back, and reload SublimeText, it fails to compile again. :(

halfnibble, 2015-11-29:
Ah ha! I have a solution that makes ST3 w/ Stino and Arduino IDE both happy.

Modify the problem platform.txt line (with the `{archive_file_path}` text) to the following:
`recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}"`

Basically, replace `{archive_file_path}` with `{build.path}/{archive_file}`

This will allow ST3 with Stino to compile, and Arduino IDE will also compile withe following warning:
>Warning: platform.txt from core 'Arduino AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.

sagmua, 2016-05-30:
works for me! Thanks!