Robot-Will/Stino #316: Startup error in base.getKeywordListFromFile(keywords_file)

andig, 2015-11-12:
Code is like this:

			if len(word_list) > 1:
				print(word_list)
				keyword_name = word_list[0]
				if len(word_list) == 3:
					keyword_type = word_list[1]
					keyword_ref = word_list[2]
				elif len(word_list) == 2:
					if 'LITERAL' in word_list[1] or 'KEYWORD' in word_list[1]:
						keyword_type = word_list[1]
						keyword_ref = ''
					else:
						keyword_type = ''
						keyword_ref = word_list[1]
				cur_keyword = Keyword(keyword_name)
				cur_keyword.setType(keyword_type)
				cur_keyword.setRef(keyword_ref)
				keyword_list.append(cur_keyword)

Wordlist has 4 entries which is not handled by the code. Error message:

    UnboundLocalError: local variable 'keyword_type' referenced before assignment

loredan, 2015-11-30:
Stumbled onto the same problem, it does cause #314, here is traceback:
```
Traceback (most recent call last):
  File "/opt/sublime_text_3/sublime_plugin.py", line 74, in reload_plugin
    m = importlib.import_module(modulename)
  File "./importlib/__init__.py", line 90, in import_module
  File "<frozen importlib._bootstrap>", line 1584, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1565, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1532, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 584, in _check_name_wrapper
  File "<frozen importlib._bootstrap>", line 1022, in load_module
  File "<frozen importlib._bootstrap>", line 1003, in load_module
  File "<frozen importlib._bootstrap>", line 560, in module_for_loader_wrapper
  File "<frozen importlib._bootstrap>", line 868, in _load_module
  File "<frozen importlib._bootstrap>", line 313, in _call_with_frames_removed
  File "/home/loredan13/.config/sublime-text-3/Packages/Arduino-like IDE/StinoStarter.py", line 12, in <module>
    from . import app
  File "/home/loredan13/.config/sublime-text-3/Packages/Arduino-like IDE/app/__init__.py", line 21, in <module>
    arduino_info = base.ArduinoInfo()
  File "/home/loredan13/.config/sublime-text-3/Packages/Arduino-like IDE/app/base.py", line 17, in __init__
    self.refresh()
  File "/home/loredan13/.config/sublime-text-3/Packages/Arduino-like IDE/app/base.py", line 24, in refresh
    self.genKeywordList()
  File "/home/loredan13/.config/sublime-text-3/Packages/Arduino-like IDE/app/base.py", line 53, in genKeywordList
    self.keyword_list += getKeywordListFromPlatform(platform)
  File "/home/loredan13/.config/sublime-text-3/Packages/Arduino-like IDE/app/base.py", line 878, in getKeywordListFromPlatform
    keyword_list += getKeywordListFromCoreFolderList(core_folder_list)
  File "/home/loredan13/.config/sublime-text-3/Packages/Arduino-like IDE/app/base.py", line 860, in getKeywordListFromCoreFolderList
    cur_keyword_list = getKeywordListFromFile(keywords_file)
  File "/home/loredan13/.config/sublime-text-3/Packages/Arduino-like IDE/app/base.py", line 849, in getKeywordListFromFile
    cur_keyword.setType(keyword_type)
UnboundLocalError: local variable 'keyword_type' referenced before assignment
```

It failes on the line `HIGH LITERAL2 Constants RESERVED_WORD_2` which does have 4 words, which is not being processed correctly
Here is a list of all the lines with 4 words:

loredan, 2015-11-30:
All of the above applies to plugin in PackageControl, master branch in this repository, as far as I can tell

andig, 2015-12-08:
> All of the above applies to plugin in PackageControl, master branch in this repository, as far as I can tell

Correct. What to do- ignore the fourth word? I have no clue what it's used for, my proposal just unbreaks the plugin but is not necessary right.

PR with this fix? Are PRs still being applied to this repo?

loredan, 2015-12-08:
Not sure. And I already moved to the active fork of this plugin, https://github.com/gepd/Stino

Robot-Will, 2015-12-08:
Thanks a lot. As there are many differnt things in Arduino above 1.6.3, this plugin do not work well. I am always finding times to rewrite this plugin and a little busy now, but I am trying to make this plugin better.

andig, 2015-12-08:
@Robot-Will @loredan I'm all for forks and stuff but it it makes matters for users quite complicated when you're not aware of those forks. Why not join forces and invite trusted committers to this repo?

At any rate I fully appreciate the efforts you've been and are putting into this project.

loredan, 2015-12-08:
@Robot-Will @andig Well, honestly, I just followed a link in README.md, which points to a fork, it's a little ahead of ST4ArduinoIDE branch. And by the way, why is there such a diffeerence between master and ST4ArduinoIDE? That's not on topic of an issue, but still, PackageControl points on master branch, which looks pretty much outdated to me

andig, 2015-12-08:
My install is pre-august, I guess the link didn't exist then. I'll move to the fork accordingly, thanks for the pointer.