Robot-Will/Stino #317: Compile error: relative compiler path from arduino IDE not handled

andig, 2015-11-12:
I'm trying to compile an esp8266 project that works in arduino ide 1.6.5. on Windows.

Problem ist that `1.6.5-r5\hardware\arduino\avr\platform.txt` defines `compiler.path` as `/bin/'. That works as `/bin/` in the IDE is apparently relative to the arduino IDE/tools folder (or one of those). Stino however treats that path as absolute and can't find the compiler therefore.

I'm able to circumvent this by undefining `compiler.path` in `compiler.py` but then get stuck at the xtensa tools again.

This issue needs a general fix for handling path variables read from Arduino IDE.

Any idea?

andig, 2015-11-12:
Update: issue fixed by defining `runtime.platform.path`as below including in the respective calls:

    def getPlatformArgs(platform_path, platform_text, args):
    	args['runtime.platform.path'] = platform_path

Please let me know if you want a PR.

FWeinb, 2015-12-07:
This would be incredible useful. :+1:

andig, 2015-12-08:
@FWeinb I have honestly no clue if this is "the right thing to do" but it's working for me. Would appreciate double checking by an Arduino build process expert.

FWeinb, 2015-12-08:
I can't even choose the ESP8266 boards from "Arduino" -> "Boards". I installed the esp8266 boards via the Board Manager from the ArduinoIDE.

nugrahawahyu, 2016-05-12:
try to install using git.

closed 2017-03-09 by Robot-Will