Robot-Will/Stino #318: ArduinoIDE-installed libraries don't work in Stino

andig, 2015-11-12:
I've installed ArduinoJson through the IDE. Compiling sketch in IDE works fine but fails in Stino:

	Creating C:\Users\Documents\Arduino_Build\esp8266-webconf\esp8266-webconf.ino.cpp.o...
	"C:\arduino-1.6.5-r5\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\arduino-1.6.5-r5\hardware\esp8266com\esp8266/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DARDUINO=165 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_  -DESP8266 "-IC:\Users\Documents\Arduino_Build\esp8266-webconf" "-IC:\htdocs\esp8266-webconf" "-IC:\arduino-1.6.5-r5\hardware\esp8266com\esp8266\tools\xtensa-lx106-elf\xtensa-lx106-elf\include" "-IC:\arduino-1.6.5-r5\hardware\esp8266com\esp8266\cores\esp8266" "-IC:\arduino-1.6.5-r5\hardware\esp8266com\esp8266\cores\esp8266\spiffs" "-IC:\arduino-1.6.5-r5\hardware\esp8266com\esp8266\variants\generic" "-IC:\arduino-1.6.5-r5\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src" "-IC:\arduino-1.6.5-r5\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\arch" "-IC:\arduino-1.6.5-r5\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\include" "-IC:\arduino-1.6.5-r5\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\lwip" "-IC:\arduino-1.6.5-r5\hardware\esp8266com\esp8266\libraries\ESP8266mDNS" "-IC:\arduino-1.6.5-r5\hardware\esp8266com\esp8266\libraries\ESP8266WebServer\src" "-IC:\arduino-1.6.5-r5\hardware\esp8266com\esp8266\libraries\ESP8266WebServer\src\detail" "-IC:\arduino-1.6.5-r5\hardware\esp8266com\esp8266\libraries\ArduinoOTA"  "C:\Users\Documents\Arduino_Build\esp8266-webconf\esp8266-webconf.ino.cpp" -o "C:\Users\Documents\Arduino_Build\esp8266-webconf\esp8266-webconf.ino.cpp.o"
	C:\Users\Documents\Arduino_Build\esp8266-webconf\esp8266-webconf.ino.cpp:19:25: fatal error: ArduinoJson.h: No such file or directory
	 #include <ArduinoJson.h>
				 ^
	compilation terminated.

Any ideas how to fix the library path for IDE-installed libraries without copying everything around?

SterlingPeet, 2016-02-04:
I also had this problem, but [Installing from beta repository](https://github.com/Robot-Will/Stino#installation) fixed it for me.

Eternyt, 2016-06-16:
see https://github.com/Robot-Will/Stino/issues/350

closed 2017-03-09 by Robot-Will