Robot-Will/Stino #319: fatal error: optiLoader.h: No such file or directory

SalvorinFex, 2015-11-13:
I just installed Stino and already had 1.6.5 r2 on my system, but getting this error.

*[Stino - Start building "Arduino"...]
[  3%] Creating C:\Users\kurtr\AppData\Local\Temp\Stino_build\Arduino\Arduino\images.cpp.o...
C:/Users/kurtr/Documents/Arduino/images.cpp:1:24: fatal error: optiLoader.h: No such file or directory

 #include "optiLoader.h"

                        ^

compilation terminated.

[Stino - Exit with error code 1.]*

Any Ideas?

MarikSx, 2015-11-14:
Rename file platform.txt to backup_platform.txt or other.
Win 7-10: \Program Files (x86)\Arduino\hardware\arduino\avr
OSX: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/

gepd, 2016-01-24:
I've pushed a fix that should solve this problem. In my tests it worked with Arduino IDE 1.6.0 - 1.6.5 - 1.6.7
Branch [New Stino](https://github.com/Robot-Will/Stino/tree/new-stino)

closed 2016-01-24 by gepd