Robot-Will/Stino #321: Are these yellow dots which appear during build errors part of stino?

jwhendy, 2015-11-29:
Greetings,

When I build a `.ino` file and have an error of some sort, I get yellow dots and highlighted lines in Sublime Text using stino:

[image]

I only use Sublime for Arduino programming so I'm not very familiar with it. In googling around, I'm finding suggestions related to "linting" and tags:

- [Linting](https://www.sublimetext.com/forum/viewtopic.php?f=2&t=7146): proposed solutions were `Edit -> Mark -> Set mark` (didn't work) and `Edit -> Tag -> Tag lint` (which I don't have in my menu).

- [Tag package option](http://stackoverflow.com/questions/17543210/what-does-the-orange-color-marker-mean-in-sublime-text2): I appear not to have/be using this package, as I only have Stino in my `Preference -> Package settings` menu.

Just trying to find out how to make these go away after a successful build. Yellow dots by errors from the last build will disappear and move to the lines containing new errors, but after a successful build, the last batch of errors continues to feature the yellow dots indefinitely. I'd like to track down the source of this and make them clear if I have a successful build.

Thanks for any suggestions!

gepd, 2016-01-24:
This yellow dots are not from stino.
At least never saw it when I tried stino in clean ST installation.

Check if you don't have any other package under the package folder

jwhendy, 2016-01-24:
I don't believe I do. Under `Preferences -> Package Settings`, I only show `Stino (Arduino Like IDE)` and `Package Control`. In the package folder, I have `bz2`, `ssl-linux`, `Stino` and `User`. In `User`, I have `Stino`and some things that look related to `Package Control` and preferences.

If you try to compile this, do you not get *any* yellow dots?
```
int x = 0;

void setup() {}

// leaving off semicolon intentionally
void loop() { x = 1 }
```

If not, then it must be something related to Arch linux's `sublime` packaging. If you do, I'll try to inquire with `ST` directly.

gepd, 2016-01-24:
I can see it, I'll search more information about this, and I'll back with anything I find.

jwhendy, 2016-01-27:
Thanks for doing that, though if you can see them it's part of ST and I can follow up with them! I'll close the issue since it's not part of stino. Thanks!

closed 2016-01-27 by jwhendy

gonzoid, 2016-01-27:
Hello,

I have exactly that same problem running ST2 with an older revision of the `new-stino` branch, which I installed manually.

@jwhendy @gepd Which branch of stino are you using?