Robot-Will/Stino #322: Verify/compile Error code 1. Permission denied (OSX)

shashwat767, 2015-11-30:
When I go to veriry/compile the code in Sublime Text 2, I get the following error : 
"
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-ar: {archive_file_path}: Permission denied

[Stino - Exit with error code 1.]
"
I also get the same error if i choose to upload the same in Sublime text. 
The code compiles and uploads without any issues via the Arduino IDE.

I am on OSX El Capitan and using the Arduino 1.6.6.

Thanks!

noobka, 2015-12-18:
The same problem with Arduino IDE 1.6.7. Any suggestions?

gepd, 2015-12-18:
By now you should use a previous version of the IDE. @Robot-Will is developing a new version of the Stino plugin, when it is released you should be able to use the latest version of the Arduino IDE without problems

samsam2310, 2015-12-19:
The problem is the permission of ttyACM0 file under /dev.
You can use root to start arduino ide or just add your user to the dailout

jwhendy, 2015-12-31:
My distro, Arch, just pushed out 1.6.7 (from 1.6.5) about a week ago, and I just ran into this. I don't understand the recommendation to add one's self to the `dialout` group. What should the permissions be?

Here's what I see:
```
$ ls -l /dev/ttyACM0 
crw-rw---- 1 root uucp 166, 0 Dec 30 12:19 /dev/ttyACM0

$ groups
lp wheel uucp lock jwhendy
```

That would suggest I have access to `ttyACM0`, wouldn't it?

I did some other digging and don't entirely understand the posts, but these seem relevant:

- [Arduino forum post](https://forum.arduino.cc/index.php?topic=358313.0)
- [git arduino issue #4114](https://github.com/arduino/Arduino/issues/4114)

I took a complete stab in the dark based on the forum post and changed the following:

**Before:**
```
$ grep archive /usr/share/arduino/hardware/arduino/avr/platform.txt
## Create archives
recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}"
```

**After:**
```
$ grep archive /usr/share/arduino/hardware/arduino/avr/platform.txt
## Create archives
recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}"
```

This allowed me to successfully upload with `stino`. Again, no idea what I'm doing, but it works for me at the moment so I'm a happy camper.

closed 2017-03-09 by Robot-Will