Robot-Will/Stino #325: content menu stino is twice

mrbbp, 2015-12-22:
have twice the content of the menu
[image]

osx 10.11.1 arduino 1.6.4

gepd, 2015-12-22:
I've tried to reproduce your issue, but I couldn't do it
Please post you console output `View > Show console`

Maybe there is something it can help to solve it.

mrbbp, 2015-12-25:
this is the error in console (sorry for late answer)

error: Error loading syntax file "Packages/Arduino-like IDE/Arduino.tmLanguage": Unable to open Packages/Arduino-like IDE/Arduino.tmLanguage

gepd, 2016-01-24:
I tried to reproduce this bug again, and didn't get it. Please try installing the plugin again, I've done some changes in the code.

mrbbp, 2016-01-24:
have to remove "Arduino like IDE" (???) and "Stino" package and reinstall after adding repo...
with lot of stange behaviour about my board list...
Stino does not seems to be compatible with 1.6.6 Arduino IDE... no board are available.
with 1.6.4, look good, all board are here.
If i change langage, and open arduino menu, it look like my shoot and a reset occur (a message window said "package stino is not here") but the menu is clean and with the good langage choice...
thanks.

gepd, 2016-01-24:
- First from the package folder remove `Stino` folder
- From the `package/User` folder remove the `stino` folder

Install the plugin as is detailed in the [home page](https://github.com/Robot-Will/Stino#installation)

I'm working a new plugin called Deviot based in the main idea of stino, if you like to test it please go to: https://github.com/gepd/Deviot

Robot-Will, 2017-03-11:
You can delete all Main.sublime-menu files in [sublime packages]/User/Stino/. Thanks.

closed 2017-03-11 by Robot-Will