Robot-Will/Stino #326: core.a No such

traBolic, 2015-12-27:
Hi, i connot compile my first simple project. Im using lastest SublimeText3 and Stino.
Is there any solution for this issue?
<pre>//firstTest.ino

int led = 13;

void setup() {
  // put your setup code here, to run once:
  pinMode(led, OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:
    digitalWrite(led, HIGH);  // turn the LED on (HIGH is the voltage level)
    delay(100);               // wait for a second
    digitalWrite(led, LOW);   // turn the LED off by making the voltage LOW
    delay(100);               // wait for a second
}
</pre>
<pre>[Stino - Start building "firstTest"...]
	[  3%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\firstTest.ino.cpp.o...
	[  6%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\hooks.c.o...
	[ 10%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\WInterrupts.c.o...
	[ 13%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\wiring.c.o...
	[ 16%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\wiring_analog.c.o...
	[ 20%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\wiring_digital.c.o...
	[ 23%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\wiring_pulse.c.o...
	[ 26%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\wiring_shift.c.o...
	[ 30%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\abi.cpp.o...
	[ 33%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\CDC.cpp.o...
	[ 36%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\HardwareSerial.cpp.o...
	[ 40%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\HardwareSerial0.cpp.o...
	[ 43%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\HardwareSerial1.cpp.o...
	[ 46%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\HardwareSerial2.cpp.o...
	[ 50%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\HardwareSerial3.cpp.o...
	[ 53%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\IPAddress.cpp.o...
	[ 56%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\main.cpp.o...
	[ 60%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\new.cpp.o...
	[ 63%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\PluggableUSB.cpp.o...
	[ 66%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\Print.cpp.o...
	[ 70%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\Stream.cpp.o...
	[ 73%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\Tone.cpp.o...
	[ 76%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\USBCore.cpp.o...
	[ 80%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\WMath.cpp.o...
	[ 83%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\WString.cpp.o...
	[ 86%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core_arduino\wiring_pulse.S.o...
	[ 90%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\core.a...
	[ 93%] Creating C:\Users\Home\AppData\Local\Temp\Stino_build\firstTest\firstTest.elf...
	avr-gcc: error: C:/Users/Home/AppData/Local/Temp/Stino_build/firstTest/core.a: No such file or directory

	[Stino - Exit with error code 1.]
</pre>

oeivanov, 2016-01-17:
https://github.com/Robot-Will/Stino/issues/315

gepd, 2016-01-23:
@traBolic I've pushed a fix for this problem. In my tests it worked with Arduino IDE 1.6.0 - 1.6.5 - 1.6.7
Branch [New Stino](https://github.com/Robot-Will/Stino/tree/new-stino)

oeivanov, 2016-01-24:
@gepd it's work fine for me with Arduino IDE 1.6.7

gepd, 2016-01-24:
Excellent, I'll close this issue, opens it if you find any related issue.

closed 2016-01-24 by gepd