Robot-Will/Stino #327: avr-g++: No such file or directory

nightwayfarers, 2015-12-31:
Trying to compile basik sketch ("blink").
Arduino IDE v 1.6.5.
Stino installed using Package Control.
Set "board" and "serial port" - leonardo, com6
Here are the errors:
avr-g++: error: '-DUSB_MANUFACTURER=Unknown': No such file or directory
avr-g++: error: '-DUSB_PRODUCT=Arduino Leonardo': No such file or directory
Does anyone have ideas why it isn't working? Any help welcome
[image]

jwhendy, 2016-01-03:
Silly question, perhaps, but what `avr` related stuff do you have installed? It might help to know if this is a fresh install (never worked) or something broke.

I've not run into this error and am not sure if this is an issue with the listed options, or if `avr-g++` simply wasn't found. On my system (arch linux), `avr-g++` is provided by the `avr-gcc` package; I checked and it's not a dependency on my system (I had to explicitly install it).

Long shot, but can you do?
```
$ which avr-g++
/usr/bin/avr-g++
```

Lastly, I wonder if #242 is related? Suggests adding quotes in `/path/to/sublime-text-*/Packages/Stino/preset/platform_sam.txt`.

nightwayfarers, 2016-01-04:
I've installed Stino first time, I didn't worked with him before. 
Also I have MinGW 3.21 installed as C\C++ compilator (gcc, g++), does it matter? I don't use it with Arduino. Also no avr-related software installed. Should I install avr-gcc? 
#242: tried to change -DUSB_PRODUCT={build.usb_product} to '-DUSB_PRODUCT={build.usb_product}' in line 46, but nothing changed.

jwhendy, 2016-01-05:
@nightwayfarers well, I'm really not sure. I removed `avr-gcc` from my computer and was able to compile a `.ino` file I had laying around just fine. I also notice that I have:

```
$ locate avr-g++
/usr/share/arduino/hardware/tools/avr/bin/avr-g++
```

That's owned by the `arduino` package (same one that provides the official IDE), so maybe it's redundant/unnecessary? I know you're using Windows, but [this page](http://playground.arduino.cc/Learning/Linux) on Linux suggests that Arduino ships with `avr-gcc`. Maybe the issue is `sublime-text` or `stino` not finding the right Arduino folder?

That makes me wonder if it's related to getting sublime/stino to find your Arduino folder? There are a bunch of issues related to that summarized in #205 . Sorry I can't help more!

tonymusor, 2016-01-20:
Rename the platform.txt file (to platform.old or something like that) in your arduinoXXX\hardware\arduino\avr directory (on windows) and then see if it works. That fixed it for me.

gepd, 2016-01-23:
Guys I've pushed a fix for this problem. In my tests it worked with Arduino IDE 1.6.0 - 1.6.5 - 1.6.7
Branch [New Stino](https://github.com/Robot-Will/Stino/tree/new-stino)

closed 2016-01-24 by gepd