Robot-Will/Stino #328: restoring of Preference.stino-setting

Eternyt, 2016-01-05:
Every time I connect my board to my mac I must reselect the port because Stino does't remember it. I search the preference file of Stino and I found a line of code that was for me: "serial_port": "/dev/cu.Bluetooth-Incoming-Port", I modify this line in "serial_port": "/dev/cu.usbmodem1421", but every time I replugged my arduino on mac the file was restored like original.

gepd, 2016-01-05:
Try this:
In file: Stino\stino\pyarduino\base\serial_listener.py
replace

```
def check_target_serial_port():
    arduino_settings = settings.get_arduino_settings()
    serial_ports = list_serial_ports()
    target_serial_port = arduino_settings.get('serial_port', 'no_serial')
    if serial_ports and not target_serial_port in serial_ports:
        target_serial_port = serial_ports[0]
        arduino_settings.set('serial_port', target_serial_port)
    return target_serial_port
```

 for this
```
def check_target_serial_port():
    arduino_settings = settings.get_arduino_settings()
    target_serial_port = arduino_settings.get('serial_port', 'no_serial')
    return target_serial_port
```

Eternyt, 2016-01-06:
I try to do this, but when I open the file that you indicate me, I don't fount
```
def check_target_serial_port():
    arduino_settings = settings.get_arduino_settings()
    serial_ports = list_serial_ports()
    target_serial_port = arduino_settings.get('serial_port', 'no_serial')
    if serial_ports and not target_serial_port in serial_ports:
        target_serial_port = serial_ports[0]
        arduino_settings.set('serial_port', target_serial_port)
    return target_serial_port

```
This is the code present in my serial_listener.py
```
//something...
@deco.singleton
class SerialListener(object):
    def __init__(self, func=None):
        self.func = func
        self.serial_list = []
        self.is_alive = False

    def start(self):
        if not self.is_alive:
            self.is_alive = True
            listener_thread = threading.Thread(target=self.update)
            listener_thread.start()

    def update(self):
        while self.is_alive:
            pre_serial_list = self.serial_list
            self.serial_list = serial_port.list_serial_ports()
            if self.serial_list != pre_serial_list:
                serial_port.check_target_serial_port()
                if self.func:
                    self.func()
            time.sleep(1)

    def stop(self):
        self.is_alive = False
```

gepd, 2016-01-06:
I'm sorry the file is `serial_port.py` in  the same location

Eternyt, 2016-01-07:
Thanks a lot! It works!

closed 2016-01-07 by Eternyt

mmeisel, 2017-07-19:
Hi, I'm having the same problem, but it seems this code has changed with Stino2017. Is there any similar fix for the latest version? Thanks!

EtnasSoft, 2017-09-10:
Same problem here... How can we change the default serial port? 
It's very annoying to do it by hand after every Upload command... :(